alutov/ESP32-R4sGate-for-Redmond #119: Не подключается к серверу MQTT

Malaysk, 2022-11-17:
Здравствуйте. Залил самую последнюю прошивку, скачанную сегодня. Все как бы работает (чайник, мультиварка), но не подключается к MQTT , все время написано - MQTT connection count / state | 0 / Disconnected
Хотел спросить, есть какие то проблемы, если имя пользователя с большой буквой? Все другие девайсы без проблем подключены и работают, только с R4sGate проблема.
[image]
Уже и перепрошивал и менял все настройки и т.д. - не подключается к серверу MQTT (находится брокер в Home Assistent)

alutov, 2022-11-17:
Могут быть проблемы с символами @ или чем-то подобным в логине или пароле.

Malaysk, 2022-11-17:
Как раз в пароле есть такой символ. Подключено около 60 девайсов к MQTT, это надо будет всю сеть переделывать из за одного символа ?

alutov, 2022-11-17:
Тогда придется немного подождать.

Malaysk, 2022-11-17:
Спасибо! Конечно подожду!

alutov, 2022-11-17:
Попробуйте файл из папки jpg.

Malaysk, 2022-11-17:
Отлично! Моментально все соединилось! Спасибо!

alutov, 2022-11-17:
Давно надо было передавать логин пароль отдельными полями, а не в общей строке "логин:пароль@сервер". Просто не думал, что многие используют эти символы в пароле.

Malaysk, 2022-11-17:
А я всю голову сломал сегодня, целый день провозился, а результата небыло. Да, знаю о такой проблеме - "логин:пароль@сервер" , не первый раз проблема возникает с спецсимволом :-) . Еще раз спасибо, что так оперативно сделали.

Malaysk, 2022-11-17:
Можно еще один вопрос в этой ветке ? Пытаюсь подключить mi smart scale 2.
Указал MAC адрес , выбрал Scan MiID.
На общей странице вот так
[image]
Встал на весы для сопряжения.....и ни в какую. Подскажите как правильно ?

alutov, 2022-11-17:
Весы не нужно подключать. Нужно активировать BLE монитор, выбрав в настройках BLE monitoring-> Passive. Затем сохраняем, далее заходим на вкладку монитора, находим там весы. В поле "Name/Type" должно быть Mi Scale. В этой строке ставим тайм-аут, скажем, 60 секунд и нажимаем Ok. Все, весы прокинутся в мктт и в устройства Home Assistant.

alutov, 2022-11-17:
Scan MiID -  это опция для поиска продукт айди чайников MiKettle. Их оказалось пока только 3. Появилась при добавлении этих чайников, потом не стал убирать, правда, недавно переименовал. И все равно с этой опцией народ пытается подключить термометры и весы Xiaomi.

Malaysk, 2022-11-18:
Спасибо за подробную информацию! Все прочитал на главной странице предварительно, но именно по весам не нашел способа подключения :-) Теперь все идеально понятно!

alutov, 2022-11-19:
Брокер поддерживает русские буквы, дальше не пробовал пока что.

closed 2022-12-23 by Malaysk