alutov/ESP32-R4sGate-for-Redmond #123: Поддержка тензодатчика с HX711

dmitry-ponamarev, 2022-12-14:
Небольшой фичреквест: Возможна ли поддержка подключения тензодатчика через HX711 для мониторинга уровня воды в чайнике по весу?

alutov, 2022-12-14:
Уже думал об этом, но пока не придумал, как это сделать красиво. Там свой протокол и, что не совсем удобно, надо 2 ноги.

dmitry-ponamarev, 2022-12-14:
Есть мысль сделать подставку под чайник, в которой и разместить сам шлюз и hx711. Подойдет не для всех случаев конечно, но в таком случае питание можно попробовать взять прямо с  чайника. Экран конечно не будет в таком случае, но думаю и не всем он нужен.

dmitry-ponamarev, 2022-12-16:
По большому счету принципиальных отличий нет, т.к. в любом случае нужно будет тарировать индивидуально.
Я себе представляю это как-то так: 
В настройках задается несколько значений

1. Значение без чайника (только подставка) - будет обозначать, что чаника на подставке нет.
2. Значение пустого чайника (условные 0 литров)
3. Значение с каким-то объемом воды (например 1 литр) - для вычисления объема воды.

ну и мониторинг текущего значения, для того, чтобы задать вышеуказанные значения
Возможно можно сделать и проще, выкинув первый пункт и мониторить отсутствие чайника на подставке по отрицательному значению объема воды.

alutov, 2022-12-22:
В папке jpg тестовая версия с поддержкой тензодатчика и hx711.

alutov, 2022-12-29:
Добавлено в версии 2022.12.29

closed 2022-12-29 by alutov