alutov/ESP32-R4sGate-for-Redmond #124: Кофеварка RCM-M1519S не выключается

notabene00, 2022-12-18:
Кофеварка не реагирует на команду выключения.
Остальное работает - крепость, блокировка, статус задержки, часы - всё "забирается".
[image]
Включение работает по `.../cmd/state -> 1`
Выключение по `.../cmd/state -> 0` - не работает

alutov, 2022-12-19:
А какая версия?

notabene00, 2022-12-19:
Version App / IDF / CPU Clock / CPU Rated | 2022.12.10 / v4.4-dirty / 160MHz / 240MHz

Изначально оно и не включалось, и не выключалось. 
Подключился через родное приложение, обновил прошивку кофеварки на последнюю доступную.
Теперь не работает только выключение.

alutov, 2022-12-19:
Нет, обновление не повлияло. Это я поломал в попытках оптимизации. Поправил и закинул в папку jpg, там всегда времянки выкладываю. Можно пробовать. Чуть поправлю еще и на днях будет новая вчерсия.

notabene00, 2022-12-19:
> Поправил

Да, проблема решена. Спасибо за большую работу :)

notabene00, 2022-12-21:
Пока этот канал коммуникации еще не закрыт, задам вопрос здесь.
Хотелось бы понять: возможно ли выставить время на кофеварке на текущей версии прошивки.
В описании не нашел подобной функциональности, мобильное приложение делает это автоматом.
Но после переподключения к ESP время слетает.
Если сейчас невозможно и возьметесь реализовать, открою отдельный issue.

alutov, 2022-12-21:
Речь о часах в кофеварке? Шлюз периодически посылает команду установки времени, если сам смог получить его с ntp сервера. В вебе шлюза время и дата отображаются правильно?

notabene00, 2022-12-21:
А, черт. 
Я всё умнодомовое от глобальной сети отрезал. 
Понял, спасибо.

alutov, 2022-12-29:
Поправлено в версии 2022.12.29.

closed 2022-12-29 by alutov