alutov/ESP32-R4sGate-for-Redmond #126: Feature Request. MQTT command for refresh picture.

exotsk, 2022-12-22:
Андрей, можно сделать галку обновление картинки по MQTT команде ? для видеодвернух звонков полезно.
И реально ли  сделать ресайз картинки чтобы вписать в экран ? У меня подхватывает картики от 480, влазит в литмит памяти. Но показывает конечно же уголок.

alutov, 2022-12-22:
По порту не могу пока сказать ничего, по масштабированию надо глянуть и вспомнить, что может  jpeg декодер. Обновление картинки возможно как локальной кнопкой, так и по  мктт, если писать в топик jpg_time любое ненулевое значение.

exotsk, 2022-12-22:
если писать в топик jpg_time любое ненулевое значение.
здесь понятно, спасибо, можно отправить любое  большое, 
подгон размера был был бы пригоден. иначе придется поднимать что-то извне, если есть ресурс, то лучше локально.

alutov, 2022-12-22:
В папку jpg залил тестовую версию с масштабированием изображения. В декабрьской версии Home Assistant появилась вохможность вывода текста. И теперь можно управлять выводом картинки из интерфейса ХА.

exotsk, 2022-12-23:
Спасибо, работает,  но масштабирует как то странно, примерно 20-25% процентов не дорастягивает по ширине и высоте картинку (480х270) и порт так же не

alutov, 2022-12-23:
Масштабирование в декодере только 1:1, 1:2, 1:4. Если ширина больше 320 1:2, больше 640 1:4. Про лапы кота как раз думал, экран должен очищаться при приходе любой команды по мктт. Можно, правда еще центровку по горизонтали сделать.

exotsk, 2022-12-23:
Про масштаб понял. Болько если экран очистить, будут белые границы ? мне кажется лапы кота лучше :)

alutov, 2022-12-23:
Залил версию с центровкой.

exotsk, 2022-12-23:
C центровкой вроде симпатично стало 👍🏻 
но порт
[image]
[image]

alutov, 2022-12-23:
Ну уже и не знаю, что еще можно сделать. Можно включить Uart Debug и посмотреть, не испортил ли шлюз строку. И еще может нужно писать https, а не http. Шлюз в этом плане деревянный, автозамены делать не умеет, куда послали, туда и идет.)  Для проверки залил на гит тестовую картинку и в этой строке указал порт: https://raw.githubusercontent.com:443/alutov/ESP32-R4sGate-for-Redmond/master/jpg/tst.jpeg
У меня работает.

exotsk, 2022-12-23:
[image]
[image]

alutov, 2022-12-23:
Может быть такое, что картинка приходит в текстовом виде, похоже на html. Браузеры этот формат пони мают и уже сами преобразуют его в картинку. За jpeg не скажу, а вот с такими png файлами встречался.

exotsk, 2022-12-23:
Вполне возможно. Ну зато исключил возможный баг для других в будущем, сервак у меня для ретрансляции данных HA по http внутри локалки. иначе с https приходится выпускать наружу много лишнего.

alutov, 2022-12-23:
Пока это в тестовой сборке. А что описывать? Если включить Hass Discovery, в r4sGate появляются поле для ввода url и поле для ввода времени обновления. И кнопка обновления картинки.

exotsk, 2022-12-23:
аа, всё понял! ну это круто, хотя у тебя итак все можно через mqtt затолкать :)

artt652, 2022-12-23:
> Ну уже и не знаю, что еще можно сделать. Можно включить Uart Debug и посмотреть, не испортил ли шлюз строку. И еще может нужно писать https, а не http. Шлюз в этом плане деревянный, автозамены делать не умеет, куда послали, туда и идет.) Для проверки залил на гит тестовую картинку и в этой строке указал порт: https://raw.githubusercontent.com:443/alutov/ESP32-R4sGate-for-Redmond/master/jpg/tst.jpeg У меня работает. Вот лог: I (1655189) R4S: Url: https://raw.githubusercontent.com:443/alutov/ESP32-R4sGate-for-Redmond/master/jpg/tst.jpeg I (1656259) R4S: HTTP_EVENT_ON_CONNECTED I (1656459) R4S: Status = 200, content_length = 9476, offset_length = 9476

Андрей, здравствуйте! Есть предложение - сделать чтобы сообщения об ошибке выводились в  нижнюю статусную строку,  а на экране оставалась загруженная ранее картинка, (в настоящий момент ошибки показываются на коте )

alutov, 2022-12-23:
При ошибке достаю кота потому, что не понятно, что до этого было на экране, но подумать можно. Вот сегодня удачно получилось с загрузкой картинок с разрешением выше 320 * 176. А раньше даже и не пытался делать, думал, что памяти однозначно не хватит. Пытался резать строку адреса при ошибке с двоеточием. При получении такой строки шлюз уходил в постоянную перезагрузку. Похоже, драйвер хттп не совсем корректно отрабатывает или я чего-то еще не учел.

alutov, 2022-12-29:
Добавлено в версии 2022.12.29.

closed 2022-12-29 by alutov