alutov/ESP32-R4sGate-for-Redmond #130: Настройка NTP

Enzokot, 2023-01-03:
Добрый день.
Возможно ли в настройке NTP указывать имя в домене `.local` вместо IP адреса?
С IP адресами работает великолепно в локалке, также работает и с внешними NTP серверами (например, `pool.ntp.org`), а вот с локальными именами наотрез отказывается от синхронизации.

alutov, 2023-01-03:
Добавил опцию "Request NTP servers from DHCP" в sdkconfig, скомпилировал и закинул в папку jpg. Другого похожего по теме ничего не нашел. Функция системная, как там реализован запрос по имени, точно не скажу.

alutov, 2023-01-03:
Есть опция получения NTP сервера по DHCP, но это похоже немного не то. А другие устройства помимо шлюза получають локальный IP адрес NTP сервера по имени?

Enzokot, 2023-01-03:
Да, к примеру хост на debian и ПК на win.
Но, как мне кажется, опция с dhcp тоже нужная в конфигурации, со стороны dhcp сервера она настраивается option 4 или option 43 (пытаюсь выяснить у СТП keenetic, какая верная)

alutov, 2023-01-03:
А в кинетике включены hosthame? У меня тоже кинетик, но старый. Имена включить вроде можно с командной строки, а вот ntp сервер поднять уже нет.

Enzokot, 2023-01-03:
Да, включены, ntp сервер также там указан

alutov, 2023-01-03:
Тогда все же стоит попробовать файл из папки jpg. Может и сработает...

Enzokot, 2023-01-04:
Опция получения сервера NTP по DHCP в интерфейс шлюза не выведена?
Чтобы она сработала, нужно оставить пустой настройку адреса сервера NTP?

alutov, 2023-01-04:
Не выведена. Вроде он должен ролучить и так. Пустой оставлять нельзя, иначе шлюз заполнит пустое поле сервером по умолчанию pool.ntp.org.

Enzokot, 2023-01-04:
Включил на Keenetic для DHCP option 42, указал в настройке NTP шлюза левый hostname, все работает, т.е. шлюз подхватил время от Keenetic.

Enzokot, 2023-01-04:
На всякий случай, все доступные опции DHCP для Keenetic описаны в статье

Enzokot, 2023-01-04:
Предлагаю комбинацию настроек: вывести флаг "NTP from DHCP", допускать пустое значение в поле адреса NTP только при включенном флаге. Если адрес NTP пустой и флаг "NTP from DHCP" снят, то в поле адреса заполнять значение по умолчанию "pool.ntp.org"

alutov, 2023-01-04:
Можно, конечно. В принципе одно другому и сейчас не мешает. У меня вроде часы подхватываются, хотя по DHCP сервера не передаются.

Enzokot, 2023-01-04:
Еще из наблюдений, если через dhcp передавать hostname, то шлюз не также как и в самой настройке NTP не может выполнить резолв имени.

Enzokot, 2023-01-04:
Я же правильно понимаю, шлюз сначала попытается взять адрес из dhcp, ели его нет или он недоступен, то будет использован адрес из параметра настройки NTP самого шлюза?

alutov, 2023-01-05:
Похоже так. Сначала с DHCP, потом из настроек. Как мне кажется, на прикладном уровне вряд ли что можно сделать с именем в локалке. Хотя, экспериментируя с получением картинки из инета, заметил, что есп в плане url более деревянная, коррекцией и переадресацией не занимается, просто идет куда послали. Помогает иногда более аккуратное указание адреса.

Enzokot, 2023-01-05:
А мы можем как то включить отладку на моем примере и попробовать поймать причину?

alutov, 2023-01-05:
Так отдадку включить можно в настройках. Другое дело, что править если что нужно системные файлы, а это грозит потерей переносимости.Может нужно добавлять к имени .local?

Enzokot, 2023-01-05:
Изначально и думал, что будет работать с .local, но пробовал все вариации, не помогает.
Что странно, настройка mqtt работает корректно. Как я понимаю, в шлюзе резолв имен для NTP и mqtt выполняет один и тот же компонент?

alutov, 2023-01-06:
Я тоже не понимаю, почему так. Вроде вызываются одни и те же процедуры. Хотя пока что не готов это утверждать, надо еще покопаться. А, скажем, картинку с этого же устройства шлюз забирает нормально?

Enzokot, 2023-01-06:
Если нет дисплея, смогу проверить?

alutov, 2023-01-06:
Не подумал.

alutov, 2023-01-31:
В папку jpg залил версию, в которой при старте добавил инициализацию mdns. 2к оперативки сожрало).

Enzokot, 2023-01-31:
В версии из папки jpg также включено получение сервера NTP из DHCP? Может можно на время отладки (или навсегда) добавить опцию отключения получения времени из DHCP? В противном случае придется постоянно отключать option 42, что влечет перезапуск службы.

alutov, 2023-01-31:
Это можно сделать. Как правильно и коротко называется эта опция? А по mdns_init тоже пока не понял, стоит ее включать и терять память или нет.

Enzokot, 2023-01-31:
mdns_init я не успел проверить, хочется все таки отключить на шлюзе опцию получения сервера NTP из DHCP.
по названию предлагаю `NTP from DHCP`

alutov, 2023-01-31:
Выключил  пока dhcp.

Enzokot, 2023-01-31:
Взял текущую версию из папки jpg, время не обновляется.
А какая сейчас частота синхронизации времени? Может первичную инициализацию NTP сделать немного позже после старта шлюза? К примеру, после подключения к WiFi?

alutov, 2023-01-31:
Так и думал. Похоже, что заявленное использование mdns почему-то не работает с ntp. Можно, конечно, резолвить вручную если в строке только буквы, но пока не знаю, сильно громоздко все получается и память жалко(

Enzokot, 2023-01-31:
Странно, что при этом резолвится корректно адрес mqtt сервера. Предлагаю больше не тратить время, оставить поле настройки и время из DHCP.

Enzokot, 2023-02-06:
Предлагаю issue закрыть или может еще какой то вариант попробуем?

closed 2023-02-07 by Enzokot