alutov/ESP32-R4sGate-for-Redmond #133: Мультиварка RMC-M92S не подключается.

KaportsevIA, 2023-01-15:
Доброго времени суток.
Решил попробовать подключить мультиварку к шлюзу. Не хочет, при том что чайник подключается влет.
А тут в статусах висит вот такое.
[image]
[image]
[image]

На мультиварке кнопку "Час" зажимал для сопряжения.

alutov, 2023-01-15:
У меня на 800 вроде таймер нажимать нужно.

KaportsevIA, 2023-01-15:
На M800 да, таймер-отсрочка, А на M92 - "Час"

KaportsevIA, 2023-01-15:
<html>
<body>

mac: ec:33:91:77:62:c4 | RMC-M92S | -77 dB | 0.0 s | 99.9 s | 09 09 52 4d 43 2d 4d 39 32 53 02 01 05 | s
-- | -- | -- | -- | -- | -- | --
  |   |   |   |   |   | 11 07 9e ca dc 24 0e e5 a9 e0 93 f3 a3 b5 01 00 40 6e

</body>
</html>

alutov, 2023-01-15:
Судя по первой картинке не проходит авторизация, стало быть кнопка не срабатывает. Тут подсказать точно не смогу, на руках такой нет. По розеткам помню, что надо держать до сигнала, а дальнейшее удержание кнопки приводит к очистке всех имеющихся привязок. Может быть нужно подобрать момент нажатия. Либо рестартануть шлюз в момент нажатия кнопки.

alutov, 2023-01-15:
Как вариант может быть нужно обновиться через родное приложение редмонда, если прошивка сильно старая. Опять же, сам не сталкивался, но, говорят, такие случаи были.

alutov, 2023-01-15:
И еще по бле монитору интересный момент. прошло 99 секунд с момента принятия последнего пакета. У меня 800 мультиварка перестает слать рекламные пакеты после соединения. Может и 92 с чем-то соединилась в итоге?

alutov, 2023-01-15:
И как вариант, можно попробовать обновить шлюз прошивкой из папки jpg.

alutov, 2023-01-17:
Поправлено в версии 2023.01.17.

alutov, 2023-01-19:
В 17 версии, похоже, авторизация тоже не проходила. Вроде как должно работать в версии 2023.01.18

alutov, 2023-01-21:
Поправлено в версии 2023.01.18.

closed 2023-01-21 by alutov