alutov/ESP32-R4sGate-for-Redmond #136: REDMOND SkyKettle G203S

Sinicin1803, 2023-01-20:
Здравствуйте Андрей, громадное спасибо за Ваш проект!

Вот такой чайник "REDMOND SkyKettle G203S" заработает с Вашим замечательным шлюзом?
А то у меня уже третий RK-G211S уже износился, хочу чтонить другое выбрать, заодно и побольше объёмом.
Либо может быть посоветуете чего другого из моделей, главное чтобы:
1) Работало с Вашим шлюзом
2) Был прозрачным и с подсветкой
3) белый корпус
4) Не меньше 1.7 объём

З.Ы. В описании строка:
Redmond SkyKettle RK-G240S / RK-G204S / RK-G210S / RK-G211S / RK-G212S / RK-G214S / RK-M216S
продублирована

alutov, 2023-01-20:
Протокол у чайников серии 200-240 вроде бы одинаковый, а потому в меню 200 и 240 это одно и то же. Потому и строка дублируется. У меня 2 чайника, 173 и 216, но они не стеклянные и подсветка у 216 так себе. В 173 нужно уже менять пластиковый указатель уровня, 216 пока вроде без нареканий.

Sinicin1803, 2023-01-20:
А по "REDMOND SkyKettle G203S" - заработает со шлюзом?

alutov, 2023-01-20:
Думаю, да.

Sinicin1803, 2023-02-12:
Подтверждаю - REDMOND SkyKettle G203S работает со шлюзом, настроил, всё работает.

Вот только не пойму, как я на старом (RK-G211S) настраивал плавную подсветку, что типа от одного цвета переходит в другой по кругу...

bvp, 2023-03-18:
> Протокол у чайников серии 200-240 вроде бы одинаковый
Немного оффтопа: а где есть информация о протоколе? Или reverse engineering? Просто у меня есть люстра от семи огней, с пультом, и хотелось бы узнать как изучить протокол?

alutov, 2023-03-19:
Я так понимаю, изучение лога по ble, андроид позволяет это делать. И еще изучение apk. По чайникам многое было сделано давно, были об этом протоколе статьи на хабре. Были и хорошие проекты, [например от olehs](https://github.com/olehs/r4sGate). Только писалось тогда все это на ардуино, а там была глючная и ограниченная одним соединением ble библиотека. А этот проект просто переписанный под esp-idf и доработанный под свое видение проект от olehs, за что ему большое спасибо.

closed 2023-05-24 by alutov