alutov/ESP32-R4sGate-for-Redmond #140: Redmond SkyKettle RK-M170S

Ranzo, 2023-01-26:
Чайник к шлюзу подключился без проблем, но в веб интерфейсе хаба не отображается температура в чайнике. Хотя в штатной приложухе температура есть. Могу прислать скриншоты, если нужно. Прошу помощи )

P.S. спасибо за этот проект =)

alutov, 2023-01-27:
Нужно еще в настройках выбрать тип чайника. В описании 171 чайник не упоминается, так как я точно не знаю, какой протокол он использует. Может быть, конечно, что протокол у него свой и шлюзом не поддерживается. Но скорее всего подойдет 170 или 173 чайник, причем, судя по описанию (у 171, кака и у 173, температура выставляется кнопками с интервалом 5 градусов, а у 170 5 или 6 фиксированных значений) скорее всего именно 173.

Ranzo, 2023-01-27:
Да, Вы были правы. Я по началу выбрал в настройках модель RK-M170S. Теперь, поменяв ее на RK-M173S все заработало в штатном режиме. Думаю, что можно добавить в поддерживаемые чайники и 171 модель.

closed 2023-01-27 by Ranzo

alutov, 2023-01-27:
Добавлю, спасибо!