alutov/ESP32-R4sGate-for-Redmond #141: Сенсор *_kettle_ontime

Enzokot, 2023-02-04:
Нome assistant версии 2023.2 после старта системы в логах предупреждение:
`2023-02-04 12:40:01.464 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.r4s01_kettle_ontime has device class None, state class total_increasing and unit None thus indicating it has a numeric value; however, it has the non-numeric value: 113:12:25 (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`
Видимо нужно передавать числовое значение или убирать `state class`

alutov, 2023-02-06:
Добавил device_class: duration. Файл в папке jpg. Вроде у меня не ругается, но нужно еще проверить. Надо сказать, в логе полно ошибок. Например, расход в m3/h и класс water несовместимы, это допустимо только для газа.

Enzokot, 2023-02-06:
Чтобы не было в *_kettle_ontime ошибки, значение должно быть int, а сейчас судя по скриншоту - string.
<details><summary>Скриншот</summary>

[image]

</details>

alutov, 2023-02-06:
Значит, уберу все атрибуты. Если ХА такой тупой и не понимает длительность в часах минутах и секундах.

Enzokot, 2023-02-06:
Атрибуты то правильные и они собираются потом в Long-term Statistics, просто система никак не сможет собрать статистику по строке https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics
Вот к примеру другой +/- аналогичный сенсор:
[image]

alutov, 2023-02-06:
Время в секундах. Для ХА хорошо, конечно, для других систем не знаю. Наглядности меньше стало.

Enzokot, 2023-02-06:
В качестве предложения, может сделать отдельное значение для ХА, например working_time_int и топик discovery перевести на числовое значение?

alutov, 2023-02-06:
Пусть уже будет в секундах.)

alutov, 2023-02-06:
Посмотрел сейчас у себя. Вроде и не ругается, но статистику отображает неправильно.

Enzokot, 2023-02-06:
У меня в логах тишина, отображается в интерфейсе корректно. В статистике предлагает сохранение. На мой взгляд идеально.
[image]

alutov, 2023-02-06:
Оставляем так.

Enzokot, 2023-02-06:
Большое спасибо за исправления!

closed 2023-02-06 by Enzokot