alutov/ESP32-R4sGate-for-Redmond #144: Кофеварка REDMOND SkyCoffee M1505S

Kuzav, 2023-03-01:
Добрый день.
Подскажите пожалуйста, можно подключить кофеварку REDMOND SkyCoffee M1505S (Coffee RCM-M1505S)?
В BLE monitor определяется хорошо. Модель и mac считывает. 
Но коннектиться не хочет. Не сопригается.
Пробовал выставить как RCM-M1519S. Не получается.

BLE found Name / MAC / RSSI | RCM-M1505S / ebbb3edaf442 / -60 dB
Coffee RCM-M1505S not connected

alutov, 2023-03-02:
Я вот не помню, что на кофеварке нужно нажимать для сопряжения. И еще устройства редмонд поддерживают одновременно только одно соединение.

Kuzav, 2023-03-03:
Приветсвую. 
Cейчас установил на телефон приложение Gateway ( Шлюз от Redmond )
Кофеварка отлично сопрягается. МАС адрес тот же, что и определил BLE monitor.
Далее вырубил этот шлюз и пробую коннектить с ESP-32.
Чтобы сопрягать кофеварку нужно нажать и удерживать более 5 секунд клавишу включения. (вообще на кофеварке всего одна кнопка)
ESP-32 отлично видит кофеварку. Верно определяет его МАС. И верно определяет NAME.
Далее в BLE Devices Setting
- вписываю МАС и указываю, что это RCM-M1519S, сохраняю, ребут...не коннектится.
- вписываю название RCM-M1505S вместо МАС и указываю, что это RCM-M1519S, сохраняю, ребут..не коннектится
- вписываю МАС и указываю, что это NOT DEFINED, сохраняю, ребут..не коннектится
Может есть еще варианты законнектить?

alutov, 2023-03-04:
Лучше всего соединять по мак адресу, копируя его из бле монитора. Двоеточия при вводе игнорируются. В принципе, для соединения достаточно выбрать любое редмонд устройство. Чтобы сказать что-то определенное, нужно соединить есп с компьютером, как и при прошивке, запустить терминальную программу, выставить в ней скорость 115200, 8 бит, 1 стоп без четности, в настройках есп отметить uart debug и посмотреть лог при попытке соединения с кофеваркой.

Kuzav, 2023-03-04:
Надеюсь, все правильно сделал.
1. в r4sGate выставил Uart Debug
2. подключил esp32 к компу
3. запустил кофеварку на сопряжение
4. запустил Монитор Порта Про
5. Выставил Порт и скорость.
6. Сохранил лог 
[log шлюз ESP32.txt](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/10887996/log.ESP32.txt)

alutov, 2023-03-04:
Все верно))) Главное вот:
[0;32mI (5343) R4S: Write_auth 1:[0m
[0;32mI (5343) R4S: 55 00 ff b6 2c 27 b3 b8 ac 5a ef aa [0m
[0;32mI (5393) R4S: Read_auth 1:[0m
[0;32mI (5393) R4S: 55 00 ff **00** aa [0m
[0;32mI (5393) R4S: Authorize 1 Redmond error[0m
Шлюз запрашивает соединение, кофеварка его отвергает, причем и в шифрованном, и нешифрованном соединении. Это уже на прикладном уровне, физика вроде бы отрабатывает нормально. Скорее всего не проходит сопряжение. Единственное, что могу сказать, шлюз пытается соединиться 3 раза в нешифрованном режиме, потом 3 раза в шифрованном и далее по кругу. Может, стоит пробовать нажимать кнопку несколько раз не рестартуя шлюз до соединения.

Kuzav, 2023-03-04:
Ок. Попробую

alutov, 2023-03-04:
Можно потом быть еще подержать кнопку еще дольше, чтобы кофеварка забыла все привязанные устройства и потом все привязать заново.

Kuzav, 2023-03-06:
Ура, удалось привязать.
Дольше держал кнопку. Может это решило проблему. 
Пробовал все, перезагружал шлюз, кофеварку. Снова сопрягал
В какой момент все законнектилось и пробросилось в НА. 
Большое спасибо за вашу работу.

closed 2023-03-19 by alutov