alutov/ESP32-R4sGate-for-Redmond #148: Создание топиков MQTT

pmaster555, 2023-04-17:
Удалил старые топики, он по идее должен создать заново, создает .. но.
[image]

если удалить и перезапустить есп, каждый раз прибавляется еще одно вложение )

alutov, 2023-04-17:
Что-то не очень похоже на баг шлюза. Топики всегда создаются в корневой папке. Может с брокером что-то не так?

pmaster555, 2023-04-17:
iobroker, шлюз там не единственный по mqtt.  но чудачит только он так..
сейчас спецом стер флешку на есп, залили прошивку заново, настроил. первый пуск норм. Если удалить status из ветки, то начинается.
[image]

pmaster555, 2023-04-17:
и еще заметил
[image]

зачем он в обе папки пишет?

alutov, 2023-04-17:
Что касается iobroker-ского встроенного брокера. Когда сам им пользовался, заметил, что при попытке записи относительно большого количвества топиков при старте шлюза некоторые прописываются, скажем так, с защитой от дальнейшей записи. Чтобы вернуть работоспособность, приходилось ручками удалять дефектные топики, чтобы затем шлюз прописал их еще раз. Это было отмечено в описании шлюза. Далее всплывали еще странности вроде замены данных вида "255,255,255" на "YYY". Ну а терпение кончилось, когда в одной из новых версий брокера было добавлено  неотключаемое автоматическое определение типа данных. Перешел на HA с mosquitto брокером, все проблемы ушли,  Возможно, стоит в брокере поиграться настройками, в частности, увеличить размер данных в топиках,  и отключить Hass Discovery. Сейчас тестировать шлюз с iobroker не имею возможности. Что касается "дублирования", это не баг, а фича. В папке r4s выводится лучший уровень сигнала метки от нескольких шлюзов и номер шлюза, принимающий этот сигнал. Подробнее в описании.

closed 2023-07-12 by alutov