alutov/ESP32-R4sGate-for-Redmond #150: Поддержка Bluetoth счётчиков (вода, газ) Элехант

egmen, 2023-04-22:
Можно ли добавить поддержку счётчиков Элехант https://elehant.ru/
Они как раз по BLE работают. Уже есть реализация протокола для Home Assistant https://github.com/SzenProgs/elehant_water/blob/master/custom_components/elehant_water/sensor.py, но когда сервер далеко, то BLE-шлюз был бы кстати.
Могу попробовать портировать решение поддержки счётчиков на шлюз, если оно к месту в этом проекте

alutov, 2023-04-22:
Я так понимаю, в интеграции считывание показаний идет из рекламных пакетов. Тогда пакеты должны быть видны в ble мониторе.

egmen, 2023-04-22:
Странно, но почему то шлюз их не видит.
Программа BLE Scanner на Android отображает 2 устройства с mac b0:01:02... и b0:04:02... И они присылают какие-то данные.
Но в BLE Monitor на R4sGate не нахожу этих устройств...

alutov, 2023-04-22:
Это смотря с каким интервалом идут пакеты. Если он больше 5 минут(300 сек), то шлюз очищает строку с данными. Ну и шлюз может пропускать пакеты, так как он не постоянно висит на рекламных каналах.

egmen, 2023-04-22:
Как я заметил счётчики отправляют показания 1 раз в минуту и только когда есть расход воды.
Много раз перезагружал шлюз (чтобы список BLE Monitor очищался) и поймал-таки пакет от одного из счётчиков
mac: b0:01:02:00:fb:aa |   | -64 dB | 0.0 s | 54.7 s | 14 ff ff ff 80 fc a0 01 02 01 aa fb 00 b8 a5 0c 00 58 78 0a 14

Значит принципиально, ничего не мешает разработать механизм расшифровки этих пакетов и отправки в MQTT. Повторюсь, что могу заняться этим сам. Если это не будет противоречить направлению развития проекта.

alutov, 2023-04-22:
Если не установить тайм-аут, то строка счетчика пропадет через 5 минут. Не думал, что все 24 позиции забиты данными, у меня занято где-то 13.  Я так понимаю, серийник aa fb 00 = 64426, расход f2 a5 0c 00 = 82.8914 m3, температура 81 0a = 26.89°C.

egmen, 2023-04-22:
У меня все 24 строки забиваются за 20 секунд.
Вы расшифровали данные корректно.

alutov, 2023-04-22:
Добавлю в шлюз.

egmen, 2023-04-22:
Ночью удалось поймать оба счётчика на BLE Monitor
Холодная вода СВД-15 (b0:01:02:00:fb:aa)
`14 ff ff ff 80 fc a0 01 02 01 aa fb 00 b8 a5 0c 00 58 78 0a 14`
`14 ff ff ff 80 08 a0 01 02 01 aa fb 00 f2 a5 0c 00 58 81 0a 14`
`14 ff ff ff 80 14 a0 01 02 01 aa fb 00 01 aa 0c 00 5b 23 0c 14`

Горячая вода СВТ-15 (b0:03:02:00:0e:79) - особенность в том что измеряет 2 расхода: вода ниже 40 градусов и выше.
`14 ff ff ff 80 7e a0 01 02 03 79 0e 00 bf 09 00 00 75 ae 15 12`
`14 ff ff ff 80 82 a0 01 02 03 79 0e 00 bf 09 00 00 75 ec 15 12`

Вот фото показаний с оригинального Bluetooth приёмника (последняя строка из BLE Monitor примерно соответствует значениям с фото):
[image]
[image]

alutov, 2023-04-23:
Можно пробовать, файл в папке jpg.

egmen, 2023-04-23:
Удалось "поймать" mac: b0:01:02:00:fb:aa
`14 ff ff ff 80 2c a0 01 02 01 aa fb 00 27 af 0c 00 5c 84 0c 14`
Показания в HA долетели! Но разнятся с экранчиком: на экране 83 м расход и серийный номер другой
[image]

egmen, 2023-04-23:
Второй попался, в пассивном режиме список заполняется медленнее. Это удобно )
СВТ-15 mac: b0:04:02:00:0e:79
`14 ff ff ff 80 88 a0 01 02 04 79 0e 00 4b 3c 04 00 75 67 16 12`
Все показания соответствуют реальным. Но нет показателя "Горячая вода ниже 40 градусов", на экранчике выше он равен 0,249м
[image]

egmen, 2023-04-23:
Интересная конфигурация СВТ-15: устройство одно, но он "светится" с двумя mac-адресами:  b0:04:02:00:0e:79 и b0:03:02:00:0e:79. Дополнительно обрабатывать эту особенность, как мне кажется, лишне.
Вот его показания, там есть искомая цифра 0,249м
mac: b0:03:02:00:0e:79
`14 ff ff ff 80 9e a0 01 02 03 79 0e 00 c0 09 00 00 75 0e 16 12`
[image]

alutov, 2023-04-23:
Чуть ошибся, поправил. Так как привязка в мониторе идет по маку, то счетчик горячей воды распадается на 2 записи. Чтобы сильно не извращаться.

egmen, 2023-04-23:
По результатам тестирования:

1. Результаты долетают до Home Assistant!
2. СВД-15 некорректные результаты (пример от b0:01:02:00:fb:aa)
3. СВТ-15 результаты корректные, прилетают от 2-х разных mac, отличающихся вторым значением b0:04:02:00:0e:79 и b0:03:02:00:0e:79

alutov, 2023-04-23:
Это последняя версия?

egmen, 2023-04-23:
В последней версии прошивки все показания, которые можно сравнить с экранчиком совпадают. Из дополнительных фишек, появилась "Температура холодной воды", класс!
[image]

alutov, 2023-04-23:
Единственное, при пропадании счетчиков по тайм ауту сохраняются последние показания до рестарта шлюза. Мониторить наличие счетчика можно по rssi. В принципе, можно при пропадании счетчика обнулять какой-то параметр, скажем, температуру. Оставляем пока так?

egmen, 2023-04-23:
1. Показания оставляем, обнулять принудительно никакой параметр не нужно.
2. Было бы удобно, чтобы после перезагрузки шлюза тоже принудительно НЕ "затирались" нулями.
3. Ещё размышления по поводу установки таймаута в интерфейсе шлюза. Счётчики передают информацию только во время использования. Устанавливать 600 сек? На что влияет таймаут для редко выходящих на связь устройств, только на датчик "Дома/Не дома"?
4. Ещё из желательных параметров - время последнего выхода на связь, но это вроде бы штатным образом видно из показаний счётчика в HA.
5. Из общих замечаний принудительный ввод mac-адресов либо масок в BLE мониторе известных устройств. Сейчас мои счётчики "закрепились" на второй странице под номерами 17, 19, 20. Было бы удобно, чтобы они были подряд в начале на первой странице.

Так что текущая реализация вполне работоспособна. Спасибо огромное за столь быструю реакцию. Буду мониторить надёжность в процессе эксплуатации.

alutov, 2023-04-23:
Шлюз пока что не запоминает все показания меток (только мак и тайм аут) в нврам чисто из экономии памяти. В текущей конфигурации ее не так много. 
Тайм аут в счетчиках влияет на датчик дома не дома и на rssi. В других устройствах (например, термометрах) полностью обнуляются данные. Ориентировался на отслеживание меток.

closed 2023-05-03 by egmen