alutov/ESP32-R4sGate-for-Redmond #161: Home Assistant 2023.8 requires different device name

slydiman, 2023-08-02:
После обновления Home Assistant на 2023.8 ругается:
```
Discovered entities with a name that starts with the device name

Это приведет к неисправностям в версии 2024.2.0. Пожалуйста, устраните эту проблему перед обновлением.
Some MQTT entities have an entity name that starts with the device name. This is not expected. 
To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. 
Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

climate.r4s01_kettle_temp
sensor.r4s01_kettle_rssi
switch.r4s01_kettle_heat
switch.r4s01_kettle_boil
sensor.r4s0_gate_rssi
button.r4s0_gate_restart
```
В Home Assistant discovery к примеру entity name `r4s01.Kettle.temp` и device name `r4s01.Kettle`. 
Нужно чтобы они не начинались одинаково. К примеру сделать device name `r4s01.Kettle.device`.

alutov, 2023-08-03:
Наверное, не прокатит(. Думаю добавить в начале имени устройства d: dr4s01.Kettle. На днях обновлюсь и проверю.

alutov, 2023-08-03:
Сделал первую букву имени устройств большой. Бинарный файл в папке jpg.

slydiman, 2023-08-03:
Работает. Предупреждение ушло.

alutov, 2023-08-04:
Это временное решение. Пересмотррю еще все имена, чтобы все в ХА было красиво.

slydiman, 2023-08-04:
Ну всего девайсов максимум 5, на практике 1-2. Мак адрес тут выглядит излишним.
Там главное чтобы имена не начинались одинаково, именно чтобы entity name не начинался с device name (целиком).
Проверяется if entity.name.startsWith(device.name).
То есть мой вариант `r4s01.Kettle.device` должен работать.
А отличие в регистре букв - не очень решение, в будущем startsWith() с игнором регистра может дать ту же проблему.

KaportsevIA, 2023-08-04:
Я имел ввиду МАК-самого ESP32-WiFiмодуля. Он точно у плат уникальный.

alutov, 2023-08-04:
Переделал имена. Для корректного измененеия сущностей желательно выбрать в настройках Delete Mqtt Topics.

Enzokot, 2023-08-10:
@alutov сейчас получается вместо `switch.r4s01_kettle_boil` остается `switch.boil`?

alutov, 2023-08-11:
Ну да. Раньше шлюз пытался делать уникальное имя, прибавляя в качестве префикса имя устройства. Сейчас HA делает это сам.

closed 2023-08-17 by alutov