alutov/ESP32-R4sGate-for-Redmond #165: парсинг Manufacture Data, счетчика газа Элехант

mitri4, 2023-08-27:
Уважаемый, Андрей!

Во-первых, спасибо за Ваш труд. Хаб на Вашей сборке успешно крутится на ESP32, безмерно радуя)

Я пытаюсь собрать простое решение для счетчиков Элехант, основываясь на ESPHome. И у меня остался последний шаг - парсинг advertise пакета, он у меня приходит в точно таком виде как и и на Вашей сборке
14.FF.FF.FF.80.AE.0A.01.01.12.77.(05.00.85).F0.31.00.6B.E0.D1.0B 
есть мнение, что искомое значение находится в тех трех байтах, что выделены скобками. Однако при переводе их в десятиричную систему появляются различия в цифрах должно быть 3272837 (судя по показаниям в фирменном приложении, а также то что показывает Ваша сборка), а по факту эти три байта = 327813, не могу понять, что я упускаю. Пытался вычленить ваши расчеты в сборке, но видимо знаний не хватает(( 

Буду рад любой подсказке. Заранее, благодарю)

alutov, 2023-08-27:
14.FF.FF.FF.80.AE.0A.01.01.12.(77.05.00).(85.F0.31.00).(6B).(E0.D1).0B
По скобкам сдева направо серийник, показания, батарея, температура.

mitri4, 2023-08-27:
Спасибо!

closed 2023-08-27 by mitri4