alutov/ESP32-R4sGate-for-Redmond #166: Xiaomi Mijia 2 как прописать

booroondook-eng, 2023-09-20:
В мануале написано, что поддерживаются термогигрометры Xiaomi Mijia 2, но не написано, как именно их подключать.
Я пошел "по наитию". Запустил BLE-монитор, нашел один из термометров (у меня их 8 штук в доме в общей сложности), узнал его MAC-адрес.
Затем перешел в настройки. Указал в первом поле этот самый MAC-адрес.... А дальше-то что делать? В выпадающем списке протоколов нет ничего, похожего ни на Mijia, ни на Xiami. Попробовал там вообще ничего не выбирать. Сохранил настройки, устройство перезагрузилось.
Теперь на второй закладке (на первой у меня чайник Редмонд) появился MAC-адрес добавленного термометра. Но при переходе на эту закладку там по-прежнему надпись "Device not defined", и на этом всё.
Скажите, что я неправильно сделал?

alutov, 2023-09-21:
Нужно в BLE мониторе найти интересующий термометр и в поле timeout установить ненулевое значение. Все. У меня тайм-аут 400 секунд. Все вроде понятно написано. В общем списке термометров нет, они есть только в BLE мониторе.

booroondook-eng, 2023-09-21:
Сделал, как вы написали. В MQTT теперь имеются данные по RSSI (-93), STATE (ON)  и gtnum (0). И всё. Данных о температуре и влажности нет.
Прошивка датчика - PVVX 4.3, advertising Type: PVVX (Custom)

alutov, 2023-09-21:
Честно говоря, не слежу за прошивками PVVX. Шлюз поддерживает 2 нешифрованных формата atc141 (один знак после запятой) и pvvx custom(2 знака после запятой). Если протокол опознан, то в стоке термометра в поле Name/Type во второй строке должно быть "LYWSD03MMC". Если этого нет, то протокол шлюзом не поддерживается.

alutov, 2023-09-21:
Проверил свой квадратный термометр. Прошивка 4.3 в режиме custom все работает.

closed 2023-09-21 by booroondook-eng