alutov/ESP32-R4sGate-for-Redmond #173: Добавление устройства Giiker Super Cube i3

ChapayTv, 2023-10-24:
Здраствуйте, хотелось бы узнать, возможно ли добавить [данное устройство](https://market.yandex.ru/product--golovolomka-xiaomi-3x3x3-giiker-super-cube-i3-chernyi-oranzhevyi/536915025?skuId=100749703734&sku=100749703734) в поддержку?
[Здесь возможно полезная информация об устройстве](https://github.com/hakatashi/giiker/tree/master) 
[image]
[image]

alutov, 2023-10-25:
Ну это устройство вряд ли можно отнести к категории умного дома.

ChapayTv, 2023-10-25:
У интергации Passive BLE  Monitor есть поддержка, но там другая модель 
[image]