alutov/ESP32-R4sGate-for-Redmond #178: LYWSD02

I-Crazy-Demon, 2023-12-11:
Здравствуйте уважаемый Андрей
Имеются вот такие часики "Xiaomi Mijia Temperature And Humidity Electronic Watch"  и вот такой проект https://github.com/alive-corpse/LYWSD02-LYWSD03MMC-MQTT с помощью которого я забирал с часиков данные. К сожалению хомассистант пришлось перенести на другой сервер, который просто не достаёт до этих часиков. 
А вот Ваш r4sGate0 находится практически рядом.
Как-то нельзя, чтобы Ваш гейт мог получать данные с этих часиков?
BLE монитор их видит, но данных, кроме мощности сигнала не показывает...
https://disk.yandex.ru/i/GIDORu0RZklG2w

alutov, 2023-12-12:
Как я понимаю, в рекламных пакетах никакой полезной информации нет. А в режиме постоянного соединения быстро сдохнет батарея. Пока не знаю, если буду переделывать шлюз в режим соединения по требованию, может и смогу добавить,но пока сильно что-то менять ради термометра не хочется. На него разве нет кастомных прошивок, чтобы температура и влажность были в рекламных пакетах?

artt652, 2023-12-13:
Вроде бы есть у них реклама. 
https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/ble_parser/xiaomi.py

alutov, 2023-12-16:
Ага, почитаю, спс. Хотя не понимаю, зачем нужны такие часы. Шлюз в принципе сам умеет и время, и дату, и градусник, и еще картинку со двора.)))

I-Crazy-Demon, 2023-12-16:
Ну так шлюз физически находится в одной комнате. И тянуть провода с датчиками по другим комнатам, сами понимаете, не совсем удобно,  а в моём случае и совсем не реально. По этому возможность подключать ble датчики от xiomi - это очень большая и важная (по крайней мере для меня) часть Вашего проекта - за что  Вам большое спасибо!
Но в ванной у меня стоят именно такие часы. Пунктик у меня такой, чтобы время можно было видеть, ну и снимать температурные данные, используя одно устройство.
Не, если эти часы нельзя будет засинкать через Ваш гейт, я просто в ванную поставлю простой ксиомишный датчик, что прекрасно работает с Вашим шлюзом.
Это так... просто хотелось бы, по возможности. Собственно как я в своё время просил и Вы потом релизовали работу с температурными датчиками ксиоми.

alutov, 2023-12-18:
Нужно несколько вариантов рекламных сообщений с часов, желательно не картинкой, а текстом. В одном он передает батарею, в другом или же температуру и влажность вместе, или же по отдельности, вопрос в этом.

I-Crazy-Demon, 2023-12-18:
А как это сделать?

alutov, 2023-12-18:
Копипастом из бле монитора. Нужно дождаться разных пакетов. Их там 2 или 3 варианта.

I-Crazy-Demon, 2023-12-18:
Столько достаточно?
Или за какой промежуток времени надо данные собрать? В ручную явно часть
пакетов пропускается мимо... Надо какую нить софтину ставить на комп
(win10) чтобы нормально собрать пакеты...
mac: e7:2e:01:73:14:7a -77 dB 17.1 s 2.0 s
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 68 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 0d 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 a4 7a 14 73 01 2e e7 09 06 10
02 08 02
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 af 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 cb 7a 14 73 01 2e e7 09 06 10
02 08 02
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 dc 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 f4 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 06 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 1b 7a 14 73 01 2e e7 09 0a 10
01 26 01
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 46 7a 14 73 01 2e e7 09 06 10
02 08 02
02 01 06 03 02 1a 18 14 16 95 fe 70 20 5b 04 51 7a 14 73 01 2e e7 09 0a 10
01 26 01

alutov, 2023-12-18:
Пока да. Вижу 2 типа пакетов, в одном батарея, в другом влажность. Нет температуры, но ее формат есть в документации.

alutov, 2023-12-18:
Можно пробовать, бинарник в папке test. Походу добавил еще и LYWSD03MMC. По идее должен ловить их и без пререшивки, разумеется, если не включено шифрование.

I-Crazy-Demon, 2023-12-18:
Проверил - нормально работает, данные снимаются!
https://disk.yandex.ru/d/BrcopLokbYe6ug
LYWSD03MMC непрошитых в наличии нет, проверить нечем.
Большое спасибо Вам!

closed 2023-12-19 by alutov