alutov/ESP32-R4sGate-for-Redmond #182: Подключение мультикухни RMK-FM41S

Anton-Ka, 2023-12-28:
При попытке подключить к мультикухне RMK-FM41S BLE Activity показывает сначала статус "Connecting RMK-M41S (Open)", потом "Connecting RMK-M41S (Auth)", потом снова "Connecting RMK-M41S (Open)". После чего мультикухня пикает и соединения не происходит.
При этом сканер её находит, определяет имя и мак-адрес.
Перепробовал все мультиварки из списка - ничего не меняется.
Сейчас настройки такие:
[image]
Вот так на главном экране:
[image]
Подскажите, как можно её подключить к шлюзу и добавить в Home Assistant?

alutov, 2023-12-28:
Вообще-то пока мультикухня шлюзом еще не поддерживается, но соединяться должна. Похоже, не прошла привязка мультикухни к шлюзу. Нужно зажать ту же кнопку, что и при сопряжении ее со смартфоном до какого-то сигнала. Точнее не подскажу.

Anton-Ka, 2023-12-28:
> Нужно зажать ту же кнопку

Помогло, Спасибо!
Есть все режимы, кроме:
- Десерты
- Экспресс
- Разогрев

alutov, 2023-12-30:
В папке test есть версия с русскими названиями программ.

Anton-Ka, 2024-01-02:
> В папке test есть версия с русскими названиями программ.

Только что закачал для проверки. Не заметил отличий. В предыдущей установленной версии названия были такими же, двойными - впереди на английском, после - на русском.
А есть ли способ прописать русские названия программ а ХА?

alutov, 2024-01-02:
Разве сейчас в ХА нет русских названий программ?

Anton-Ka, 2024-01-02:
> Разве сейчас в ХА нет русских названий программ?

Ну они комбинированные.
[image]
Я что-то подумал, что через локализацию будет полный перевод.

closed 2024-01-04 by alutov