alutov/ESP32-R4sGate-for-Redmond #183: G200S теряет соединение

mistikdonbass, 2023-12-29:
Доброго дня!
В последнее время чайник стал отваливается от шлюза.
Снача грешил на блок питание ESP и дальность расположения, заменил блок питания и перенёс Шлюз в непосредственно близость к чайник.
Что можно попробовать сделать?

alutov, 2023-12-30:
И после отвала чайник со шлюзом не соединяется?

mistikdonbass, 2023-12-30:
> И после отвала чайник со шлюзом не соединяется?
Сам нет, даже если снять с подставки и поставить обратно.

alutov, 2023-12-30:
Можно попробовать выбрать в шлюзе Format NVS, потом перепривязать чайник к шлюзу по новой. Может, чайник соединяется со смартфоном, потому и шлюз его не видит?

mistikdonbass, 2023-12-30:
> выбрать в шлюзе Format NVS,

Попробую.
Это может происходить два раза за день, сегодня вот ни разу. 

> Может, чайник соединяется со смартфоном

Исключено, приложение даже не устанавливалось.

 Спасибо!

alutov, 2023-12-31:
И еще можно зажать на чайнике кнопу и держать до второго сигнала, чтобы чайник забыл шлюз.
А вообще что шлюз в вебке показывает? Каккие уровни по ble, wifi? Пытается соединиться? Останавливается на авторизации? Еще можно в настройках включить uart debug и смотреть логи по юсб. Тогда можно уже что-то определенное сказать.

mistikdonbass, 2024-01-01:
Вот момент, когда нет соединения:
[image]

После перезапуска шлюза:
[image] 

Галка Format NVS после сохранения не сохраняется.

alutov, 2024-01-01:
И не должна сохраняться. Можно попробовать файл версии esp-idf 5.0.5 из папки test.

alutov, 2024-01-05:
Что-то поменялось с другой версией? Похоже проблема именно в версии 2023.12.06, а в остальных версиях проявляется олько если установлен флаг "Disconnect BLE if ni Wifi". Поправлю в следующей версии.

mistikdonbass, 2024-01-05:
> Что-то поменялось с другой версией?

Я только сегодня обновил, ещё наблюдаю.
Хотел предложить добавить возможность перезагрузки шлюза, через определённое время.

mistikdonbass, 2024-01-12:
В версии 2023.12.23, работает стабильно, не было замечено потери связи.
Спасибо!!!

Есть два вопроса:
1. Есть закономерность, ровно в 05:45 выключает подсветку чайника (использую как ночник), и не важно есть ли автоматизация в HA или нет.

2. Есть ли возможность обнулить историю чайника, скажем Общее энергопотребление?

alutov, 2024-01-12:
1. Может быть, что-то сохранилось в памяти чайника. Можно попробовать зажать кнопку питания до упора, чтобы он забыл все настройки. Потом привязать к шлюзу еще раз.
2. Может и есть, но ничего подсказать не могу, сам не знаю.((

cyberpitroff, 2024-01-29:
Та же история, чайник RK-M216S, профиль указал RK-G200S. 
Стабильного соединения нет - отключается, обратного подключения не происходит, хотя в BLE monitor светится. 
Когда сбрасываешь все настройки - коннектиться, управляется, но потом в какой-то момент теряет соединение без восстановления.

alutov, 2024-01-30:
Конкретной информации нет, что-то советовать сложно. Интересует информация Ble activity на главной странице. Сканирование только или же шлюз все же пытается соединиться с чайником? Должно быть Scannig -> Open(...) -> Auth(...), потом уже устанавливается соединение. Если нет Open(...) - имя или мак адрес введены с ошибкой, если останавливается на Open(...) - слабый уровень, если на  Auth(...) - проблемы с пайрингом. Что значит "сбрасываешь все настройки"?

alutov, 2024-01-30:
В вебке кроме того, что писал, есть еще last reset reason. В идеале там должно быть power_on после включения питания или же software после программного рестарта. Если там проскакивает brownout, то это значит 3.3 вольта проседают под нагрузкой до 2-х с хвостиком. Ну и еще можно глянуть уровни сигнала как ble, так и wifi.

cyberpitroff, 2024-02-07:
Рапортую - дело было в расстоянии, перенес шлюз ближе (исключил одну несущую стену) - стало сильно стабильнее. 
Спасибо за софт, отличный!