alutov/ESP32-R4sGate-for-Redmond #184: Прошивка ESP32 (Atom Lite)/ESP32-S3 (AtomS3 Lite)

esik-droid, 2023-12-31:
Приветствую.
Не получается прошить ESP.
Порты видит.
Шью через flash_download_tool_3.9.5. Ни fr4sGate и fqr4sGate не прокатывает.

test offset :  0 0x0
case ok
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 1775744 bytes to 1049647...

[image]

Так же пробовал ESPHome-Flasher-1.4.0-Windows-x64 - не прокатывает

Через Easy Loder вроде и чистится и что то прошивается
[image]

Через M5Burner так же прошивается нормально (правда не совсем понятно что заливается)

Помогите что нет так. На руках оба устройства, как сделать шлюз уже не знаю...
Делаю по этой инструкции https://github.com/alutov/ESP32-R4sGate-for-Redmond?tab=readme-ov-file

alutov, 2023-12-31:
Atom Lite сегодня пробовал прошивать скачанной с гита прошивкой fr4sFate.bin с помошью flash_download tool 3.6.5. Все работает. Прошивки под ESP32S3 на гите нет, только под ESP32C3.

mistikdonbass, 2023-12-31:
Скорей всего ваше железо не будет работать.
[image]
Вот эти два не заработали.
Смотри SerialMonitor (скачать) у меня постоянный перезапуск ESP

artt652, 2023-12-31:
> > Atom Lite сегодня попробовал прошить скачанной с гитовой прошивкой fr4sFate.bin с помошью flash_download Tool 3.6.5. Все работает. Прошивки под ESP32S3 на сайте нет, только под ESP32C3.
> 
> Может что то не так нажимаю. Первый раз что то шью. Или может что то нажимать надо на самом устройстве при прошивке? Можете описать всю последовательность, пожалуйста?

Попробуйте перед прошивкой зажать кнопку на 2 секунды.
[image]

artt652, 2023-12-31:
> Скорей всего ваше железо не будет работать. ![IMG_20231229_213634](https://private-user-images.githubusercontent.com/135098430/293514327-c2c8f784-415c-415a-beb6-f9f9131a00bf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQwNjU3NTYsIm5iZiI6MTcwNDA2NTQ1NiwicGF0aCI6Ii8xMzUwOTg0MzAvMjkzNTE0MzI3LWMyYzhmNzg0LTQxNWMtNDE1YS1iZWI2LWY5ZjkxMzFhMDBiZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjMxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIzMVQyMzMwNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZThiN2EwNWQ0MDJjYTcwMjUwOTZlYmVmODMyOGM2NjdlZjkyNDkyYjk5ZGVlOGRiODdiODZmOTY1ZDkwMGQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FFDJ7hDODUSz24igk56s2udOfDwDPdzB54MJi6RhB7s) Вот эти два не заработали. Смотри SerialMonitor (скачать) у меня постоянный перезапуск ESP

Для прошивки таких, надо обе кнопки перед подключением зажимать.

mistikdonbass, 2024-01-01:
> Для прошивки таких, надо обе кнопки перед подключением зажимать.

Интересно, поподробней можно?

Попробовал одновременно зажать и прошить, по монитору после прошивки r4sGate.bin в режиме DIO:

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57`

После прошивки fr4sGate.bin в QIO:

` rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6864
load:0x40078000,len:14412
load:0x40080400,len:3668
entry 0x40080674
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6864
load:0x40078000,len:14412
load:0x40080400,len:3668
entry 0x40080674
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6864
load:0x40078000,len:14412
load:0x40080400,len:3668
entry 0x40080674
ets Jun  8 2016 00:22:57 `

На них, мне вообще ничего не удалось зашить, изначально пытался Wled.
Взял другой и шьётся туда всё.

artt652, 2024-01-01:
> Интересно, поподробней можно?
>
> Попробовал одновременно зажать и прошить, по монитору после прошивки r4sGate.bin в режиме DIO:
>
> `rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) flash read [...]
...
> 
> На них, мне вообще ничего не удалось зашить, изначально пытался Wled. Взял другой и шьётся туда всё.

#
Это стандартное для таких модулей,  зажимаешь две кнопки, сначала en , потом  boot,  переводят есп32 в режим загрузки прошивки, я зажимал обе кнопки ещё до подключения usb к компьютеру, как устройство определилось на ПК, кнопки отпускал. И старт прошивки.

mistikdonbass, 2024-01-01:
> Это стандартное для таких модулей, зажимаешь две кнопки, сначала en , потом boot, переводят есп32 в режим загрузки прошивки, я зажимал обе кнопки ещё до подключения usb к компьютеру, как устройство определилось на ПК, кнопки отпускал. И старт прошивки.

В режим boot они переводятся, но после прошивки Esp не стартует, постоянный перезапуск (как выше в логах). 
Что не заливаешь туда - не работает. Вопрос почему?

artt652, 2024-01-01:
Попробуйте отформатировать есп, и/или   прошивку пустышку залить, а потом прошиваться.

alutov, 2024-01-02:
Так и не понял, что за атом шьется. Обычный атом прошивается прошивкой для esp32 и работает. Поддержки Atom S3 в проекте нет. Для него, возможно (я не разбирался с s3), нужно править исходники и собирать отдельно в режиме esp32s3.

esik-droid, 2024-01-02:
Изначально приобрел версию S3. Пробовал с ней. Понял что она не подходит и купил обычный Atom Lite. Сейчас все попытки прошить делаю с Atom Lite. перепробовал разные варианты нажимать кнопки и вместе и по отдельности и при заливки и во время. Всегда одна ошибка
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
NO XMC flash detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Сейчас пробую через flash_download_tools_v3.6.5

[image]
Это нормально? В командной строке ничего не бежит. Но движется полоска загрузки. В конце появляется FINISH но в командной троке ничего нет вроде того что на скриншоте

esik-droid, 2024-01-02:
На ESP ничего не светится. Как проверить что все записалось?

esik-droid, 2024-01-02:
[image]

esik-droid, 2024-01-02:
Сейчас попробовал проверить через точку доступа, оказывается устройство прошилось. Спасибо всем. Буду разбираться дальше....

closed 2024-01-04 by alutov

strannick-ru, 2024-01-21:
> Ну и отлично. Будет время, познакомлюсь поближе с S3, раз на ней атомы делают.

Могу выслать Atom S3 Lite для тестов. Всё рано без дела лежит  : )