alutov/ESP32-R4sGate-for-Redmond #185: Подключение LYWSD03MMC

MrFikus13, 2024-01-01:
Доброго дня/вечера ! 
Попытался подключить данный термостат , однако столкнулся с проблемой того , что данные с него не отображаются ни в веб интерфейсе ни в HA через mqtt . Можете пожалуйста описать процесс его подключения, потому что мне кажется , что я где-то ошибся

I-Crazy-Demon, 2024-01-01:
Я только на шлюзе в BLE мониторе прописал частоту опроса и всё. Ну и
прошивку накатил с поддержкой.

artt652, 2024-01-01:
Какая именно то у вас прошивка, pvvx? Скорее всего нужно через флешер вручную выставить формат рекламных пакетов "custom" 
[image]

I-Crazy-Demon, 2024-01-01:
Извиняюсь, со слепу не рассмотрел модель. Думал о LYWSD02 речь.

Ну судя по инструкции последняя прошивка уже поддерживает LYWSD03MMC с офф. прошивкой. 
Видимо надо обновиться для начала (Вы не указали какая прошивка у Вас установлена)

"Текущая версия 2023.12.23 для [ESP32](https://github.com/alutov/ESP32-R4sGate-for-Redmond/raw/master/build/r4sGate.bin) и [ESP32C3](https://github.com/alutov/ESP32-R4sGate-for-Redmond/raw/master/C3/build/r4sGate.bin).
2023.12.23. В BLE монитор шлюза добавлена поддержка LYWSD02 часов с термометром и LYWSD03MMC термометров с оригинальной прошивкой. Ключи для LYWSD03MMC версии 2.1.1_0159 брал из [облака Xiaomi](https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor?tab=readme-ov-file)."

alutov, 2024-01-01:
Изначально LYWSD03MMC, похоже, шлет рекламные пакеты без данных. Чтобы данные появились, я заводил этот термометр в михоум. После этого в бле мониторе шлюза во второй строке имени должно появиться " LYWSD03MMC",а в поле тайм-аута должна появиться надпись "Key". Далее я брал длинный ключ  (BLE KEY) из облака сяоми и вводил в поле "Timeout etc" после чего шлюз даает ввести тайм аут, ну и даллее Ок. Это все работает на последней версии прошивки термометра, той, которая не перешивается по воздуху на кастом. Может быть, можно как-то обойтись без михоума, тут точно не скажу.

MrFikus13, 2024-01-01:
В общем вернулся я на оригинальную прошивку и сделал по вашей инструкции , единственное надпись "key" никак не появилась , поэтому попробовал подключить через сайт для прошивки и соответственно ключ взял оттуда . Однако датчик появился только в HA но абсолютно все значения равны нулю, уже не знаю что пробовать делать дальше

MrFikus13, 2024-01-01:
Проблема решилась выниманием батарейки из термостата. Единственное странно что сейчас, т.к. это не первая моя попытка

artt652, 2024-01-01:
Чтобы нулей не было , тайм-аут больше ставить надо. Минимум 300, а лучше 600 сек

MrFikus13, 2024-01-01:
> Чтобы нулей не было , тайм-аут больше ставить надо. Минимум 300, а лучше 600 сек

Поставил 600 , подождал , всё равно нули

alutov, 2024-01-01:
Дело в том, что при прошивке на кастом в шлюзе термометр будет отдельной строкой. При привязке к михоуму и появлении данных в рекламе тоже.

artt652, 2024-01-01:
Возможно сигнал слабый. Поднесите датчик  ближе к шлюзу или  можно попробовать ещё увеличить в 2 раза .

MrFikus13, 2024-01-01:
> Возможно сигнал слабый. Поднесите датчик ближе к шлюзу или можно попробовать ещё увеличить в 2 раза .

Дело точно не в сигнале , датчик и шлюз всё это время лежат почти вплотную, плюс у меня esp с внешней антенной

alutov, 2024-01-01:
Сейчас термометр на какой прошивке?

MrFikus13, 2024-01-01:
> Сейчас термометр на какой прошивке?
atc1441

alutov, 2024-01-01:
Значит, нужно поставить ноль в тайм-ауте термометра, где нули и нажать ок. Затем найти строку, где вторая строка в поле имени ATC/PVVX и установить тайм-аут 400 + Ок.

MrFikus13, 2024-01-01:
> Значит, нужно поставить ноль в тайм-ауте термометра, где нули и нажать ок.

Не уверен что правильно понял. Однако поставил тайм-аут 500 и по прошествии 5 минут опять нули. Я глянул логи и понял что они появляются раз через раз и промежуток как раз меняется в зависимости от тайм-аута

MrFikus13, 2024-01-01:
Перепрошил заново всю цепочку (датчик и esp) и нули пропали , хотя настроил как и было до этого (по скринам)

closed 2024-01-01 by MrFikus13