alutov/ESP32-R4sGate-for-Redmond #186: Установка на Xiaomi Gateway

QuAzI, 2024-01-04:
Добрый день. Есть Xiaomi Multimode Gateway ZNDMWG03LM, хочется отвязать его от облака. И есть чайник Redmond SkyKettle RK-M216S, который через [Redmond SkyKettle integration](https://github.com/ClusterM/skykettle-ha) и свисток Bluetooth 4 из Home Assistant регулярно отваливается (плюс какие-то невнятные грабли с Zigbee, не то из-за самого BT, не то из-за того что его стэк глючит и сыплет ошибки), через свисток BT5+ даже не пейрится, а через ESPHome Bluetooth Proxy доступен только почитать ([потому что только BLE](https://esphome.io/components/bluetooth_proxy.html), а с BLE мой чайник даёт только статус посмотреть).
Можно ли на этот гейтвей зашить в качестве альтернативной прошивки ESP32-R4sGate-for-Redmond ?

alutov, 2024-01-04:
Проект написан под ESP32 и в среде разработки для ESP32. Вряд ли его возможно перенести куда-либо, наверное проще написать все с нуля.

QuAzI, 2024-01-04:
По какой-то непонятной мне причине мой M216s каждый раз уходит в оффлайн при попытке им порулять через BLE и вообще через BT5+

artt652, 2024-01-04:
В шлюзе от Xiaomi используется другое железо, эта прошивка для него не подойдёт.

closed 2024-01-04 by QuAzI