alutov/ESP32-R4sGate-for-Redmond #187: Установка ESP32-R4sGate-for-Redmond из ESPHome

QuAzI, 2024-01-04:
Возможно ли использовать ESPHome для установки и обновления прошивки ESP32-R4sGate-for-Redmond?

alutov, 2024-01-04:
Не знаю, с ESPHome не работал. Залить в ESP прошивку наверное можно, но зачем ее обновлять через ESPHome, если это можно сделать из веб интерфейса.

QuAzI, 2024-01-04:
Через ESPHome удобно прописывать, какие версии сборочного энвайрмента нужны, всякие настройки (в т.ч. подключения к WiFi и Fallback WiFi), логи, обновления по воздуху без бубнов. А главное понятно как наливать, потому что на WROOM32 я смог только через ESPHome-Flasher-1.4.0-Windows-x64.exe залиться

alutov, 2024-01-04:
Тут все настраивается через веб интерфейс. И проект никак не завязан на ESPHome, а собирается прямо в esp-idf. Плюс готовые бинарники, чтобы на первом этапе не застопориться на компиляции. Ну и стандартные прошивальшики тоже вроде нормально шьют.

artt652, 2024-01-04:
Для тех кто любит yaml и ESPHome есть ещё одна [прошивка](https://github.com/KomX/ESPHome-Ready4Sky/wiki), но это уже другой проект.

closed 2024-01-04 by QuAzI