alutov/ESP32-R4sGate-for-Redmond #189: MQTT WebSocket vs classic mode

QuAzI, 2024-01-04:
Есть ли какие-нибудь рекомендации или замеры, какой из двух режимов подключения к брокеру лучше, какой менее требователен к ресурсам микроконтроллера или лучше переживает пропадание пакетов/потерю связи из-за лютых тормозов? Сейчас с одним единственным чайником ESP32-WROOM-32 регулярно оффлайн, даже перепейрить чайник обратно проблема

alutov, 2024-01-04:
Нет, сравнение режимов я не проводил, и рекомендаций дать не могу. Не пойму, что вылетает в офлайн, чайник или сам шлюз? Если чайник, причем тут мктт? Перепейрить чайник со шлюзом можно и без мктт. Вообще нужно смотреть уровни по wifi и ble. При -80dBm и меньше по wifi  esp уже начинает тормозить. По ble получше, при -90dBm держит, но с разрывами. В вебке шлюза есть соответствующие счетчики, в первом приближении можно понять, где проблема. Сам давно решил этот вопрос, используя есп с внешней антенной. Уровни поднимаются где-то на 15-20dBm. Эта рекомендация есть в описании шлюза.

QuAzI, 2024-01-04:
Вылетает ESP32 само собой, чайник рядом. У неё жесточайше начинает тормозить интерфейс, отрисовываться рывками, а потом и вовсе пропадает с таймаутом. Возможно случается флуд при опросе устройств (в эфире их троху прилично видно) или активном обновлении MQTT. Не похоже что проблемы с уровнем сигнала. Я попробовал [KomX/ESPHome-Ready4Sky](https://github.com/KomX/ESPHome-Ready4Sky) и с ним прям норм работает, только фич меньше и нет взаимодействия с MQTT брокером. Ну и вообще он пока не bt-proxy а работает практически с одной железкой эксклюзивно (один ESP = один чайник, в т.ч. в HASS)

closed 2024-01-04 by alutov