alutov/ESP32-R4sGate-for-Redmond #191: devices.types.cooking.kettle

bearklake, 2024-01-14:
А почему бы не прописать сущности, параметры и прочее как прописано для УДЯ?
https://yandex.ru/dev/dialogs/smart-home/doc/concepts/device-type-cooking-kettle.html#device-type-cooking-kettle

alutov, 2024-01-14:
Как я понимаю, типы устройств для передачи в яндекс определяет система умного дома, ХА в частности. Можно только выбрать из существующих. По Мктт раньше у него не было таких устройств, как чайник и мультиварка, наиболее подходил термостат. Сейчас вроде бы появился нагреватель. Но там так и не сделаны режимы подогрева или кипячения. Вместо этого там газ, эко, электричество и т.л. Смысла переделки под нагреватель пока не увидел. Возможно, что-то упустил, еще почитаю.

artt652, 2024-01-14:
Дело в том, что имеется разница между тем что умеет ХА и что умеет УДЯ. Основная проблема что такой сущности как чайник нет в ХА, поэтому и приходится собирать вручную в yaml. Посмотрите вики ЯХА, там  есть примеры, как правильно собрать и пробросить чайник в УДЯ.

closed 2024-01-20 by alutov