alutov/ESP32-R4sGate-for-Redmond #193: Интеграция

bearklake, 2024-01-22:
Возможно ли публиковать так в MQTT для закидывания в УДЯ через homed-cloud:

https://github.com/u236/homed-server-cloud
https://dialogs.yandex.ru/store/skills/5e05d82c-homed
https://t.me/homed_zigbee/69310
> Добавил на сервере УДЯ возможность добавления кастомных устройств. Для этого на брокере, к которому подключен homed-cloud, нужно создать следующую иерархию топиков:

1. список устройств (должен содержать ВСЕ кастомные устройства), поле id не является обязательным, в случае его отсутствия будет использоваться name:
homed/status/custom (retain)
{
  "devices": [
    {
      "name": "Мое устройство",
      "id": "myDevice1"
    },
    {
      "name": "Еще одно устройство",
      "id": "otherDevice"
    }
  ]
}

2. доступность устройства (online - доступно, любое другое значение - недоступно, можно использовать как will-топик):
homed/device/custom/myDevice1 (retain)
{
  "status": "online" 
}

3. умения устройства:
homed/expose/custom/myDevice1 (retain)
{
  "common": {
    "items": [
      "switch",
      "temperature"
    ]
  }
}

4. данные ОТ устройства (улетает в УДЯ):
homed/fd/custom/myDevice1 (NOT retain)
{
  "status": "on",
  "temperature": 22.3
}

5. данные НА устройство (прилетает от УДЯ):
homed/td/custom/myDevice1 (NOT retain)
{
  "status": "off"
}

Топики expose, fd и td разжевывать не буду, смотрите как ведет себя служба зигби, но если будет совсем непонятно - помогу. В данный момент будут работать только те типы устройств, которые работают для службы зигби, всякие бризеры и прочие синхрофазатроны нужно добавлять на сервер, мне пока некогда разбираться.

alutov, 2024-01-22:
Я так понимаю, сами топики трогать не обязательно, нужно иначе прописыать только дискавери топики. В принципе, место пока есть) только мне нужно немного времени изучить матчасть)))

alutov, 2024-02-19:
Проект в части добавления устройств пока что несколько сыроват. Автоматическое добавление устройств как я понял, пока невозможно.  Замечу, что шлюз можно напрямую подключить к алисе (сам этим не занимался, но инструкцию на 4пда видел). Подождем.)

closed 2024-03-03 by alutov