alutov/ESP32-R4sGate-for-Redmond #200: Не доступен шлюз

mistikdonbass, 2024-02-06:
Добрый день!
На версии 2024.01.14 Шлюз засыпал пару раз в неделю, на 2024.01.22 через 10 мин после перезапуска. 
На 2024.02.05 долго не игрался, как только он стал не доступен - вернулся на ...14

alutov, 2024-02-06:
Отваливается по wifi? Тогда стоит для начала обновиться ло версии 2024.02.05.

mistikdonbass, 2024-02-06:
Да, пропадает из сети Wi-Fi, с сетью всё в порядке, с питанием на 100% не могу утверждать т.к. всё понятно кем сделано, но грешить пока не могу на это.
На сейчас, Esp прошлил пустой прошивкой и Прошил я заново на последний версию - понаблюдаю.

У меня к вам одно предложение, дать возможно для переименования шлюза, у меня большая сеть WiFi и когда два шлюза в сети, то конфликт имём.

alutov, 2024-02-07:
В имени шлюза есть его номер. Если номера разные, то и имена будут отличаться. Что касается перепрошивки, можно еще пробовать Format NVS, тоже должна бы очищат информацию в том числе по wifi. Если дома меш, стоит разрешить шлюзу все точки. На последней версии он должен сам выбирать лучший уровень. На ранних версиях да, поключался к первой найденной точке доступа.

mistikdonbass, 2024-02-07:
С 00 часов 38 сообщений о не доступности.

По поводу имён, о каких цифрах в имены вы говорите?
Сейчас так "r4s3Gate"
На момент когда было два шлюза на esp32 не с3, я точно не помню, сейчас один стоит с3, было ли в имени "3". Но имя было точно такое же и виден был только тот кто первый подключился. По этому и запросил возможность о смене имени.

alutov, 2024-02-07:
Для подкключения нескольких шлюзов к одному брокеру нужно, чтобы они чем-то отличались. Для этого в настройках есть номер шлюза "Gate Number". Если подключить шлюза одним и тем же номером к одному брокеру, то каждый из них будет отваливаться. Номер шлюза есть в имени mqtt клиента и host имени в роутере. Если в кинетике зарегистрировать шлюз, а потом сменить его номер, то переименовывать шлюз придется вручную. Или же удалить регистрацию и повторно зарегистрировать шлюз.

mistikdonbass, 2024-02-08:
С именами разобрался, видемо на тот момент брал прошивку из вкладки релизы.
Но всё равно сразу не понятно за что какой Шлюз отвечает)))

Что бы исключить проблему с недоступностью шлюза по питанию, заменил БП. На 2024.01.14 отвалом меньше, по сравнению с 2024.02.05.
[image]
Весь день на 02.05 был засыпан сообщениями о не доступности.

alutov, 2024-02-19:
Посмотрел картинки [в другой теме](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/183). Шлюз проработал 6 часов, последний рестарт софтовый(проблем с питанием нет), по wifi счетчик соединений 2(как и mqtt, то есть проблема не в mqtt) и уровень -87dbm.  По паспорту в режиме g и n чувствительность esp32 -93dbm. То есть шлюз имеет маленький запас по уровню wifi, и стоит просто пройтись между шлюзом и роутером, чтобы линк оборвался. В последних версиях шлюза при недоступности шлюза при обрыве wifi и mqtt соединений в ХА становятся недоступными все подключенные к шлюзу устройства. Потому на новых версиях отвалов больше. Учитывая, что и ble  соединение с чайником имеет примерно такой-же уровень, стоит использовать вариант esp с внешней антенной. У меня при переходе на вариант в внешней антенной уровень wifi и ble поднялся примерно на 15dbm.

mistikdonbass, 2024-02-19:
Те скрин когда Шлюз был далеко.
Вот новые.
[image]
На установленной версии электричество чаще пропадает, чем шлюз)
2024.02.05 - постоянный отвал, нужно ещё успеть перепрошится назад.
Последующии не проверял.

alutov, 2024-02-19:
Ну наконец-то картинки.) В папке test новая версия. Должна помочь.