alutov/ESP32-R4sGate-for-Redmond #202: Не запускается шлюз esp32wroom

nightmare04, 2024-02-12:
Здраствуйте, пробовал прошивать и в dio и в qio, разные версии, результат примерно одинаков, плата обычная esp32wroom с алиекспресс, при прошивке ни на что не ругается, вот лог загрузки

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:6864
> load:0x40078000,len:14412
> load:0x40080400,len:3668
> entry 0x40080674
> I (27) boot: ESP-IDF v4.4-dev-4-g73db14240 2nd stage bootloader
> I (27) boot: compile time 20:50:27
> I (27) boot: chip revision: 3
> I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
> I (38) boot.esp32: SPI Speed      : 40MHz
> I (43) boot.esp32: SPI Mode       : DIO
> I (47) boot.esp32: SPI Flash Size : 4MB
> I (52) boot: Enabling RNG early entropy source...
> I (58) boot: Partition Table:
> I (61) boot: ## Label            Usage          Type ST Offset   Length
> I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (76) boot:  1 phy_init         RF data          01 01 0000d000 00001000
> I (83) boot:  2 otadata          OTA data         01 00 0000e000 00002000
> I (91) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
> I (98) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
> I (106) boot:  5 spiffs           Unknown data     01 82 003d0000 00030000
> I (113) boot: End of partition table
> I (118) boot_comm: chip revision: 3, min. application chip revision: 0
> I (125) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=61a00h (399872) map
> I (274) esp_image: segment 1: paddr=00071a28 vaddr=3ffbdb60 size=04e08h ( 19976) load
> I (282) esp_image: segment 2: paddr=00076838 vaddr=40080000 size=097e0h ( 38880) load
> I (298) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=119678h (1152632) map
> I (704) esp_image: segment 4: paddr=001996a0 vaddr=400897e0 size=1490ch ( 84236) load
> I (754) boot: Loaded app from partition at offset 0x10000
> I (754) boot: Disabling RNG early entropy source...
> I (766) cpu_start: Multicore app
> I (774) cpu_start: Pro cpu start user code
> I (775) cpu_start: cpu freq: 160000000 Hz
> I (775) cpu_start: Application information:
> I (778) cpu_start: Project name:     r4sGate
> I (783) cpu_start: App version:      1
> I (787) cpu_start: Compile time:     Feb 10 2024 07:45:44
> I (793) cpu_start: ELF file SHA256:  0ceae1032fa29d2a...
> I (799) cpu_start: ESP-IDF:          v5.2-rc1-dirty
> I (805) cpu_start: Min chip rev:     v0.0
> I (809) cpu_start: Max chip rev:     v3.99
> I (814) cpu_start: Chip rev:         v3.1
> I (819) heap_init: Initializing. RAM available for dynamic allocation:
> I (826) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
> I (832) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
> I (838) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
> I (844) heap_init: At 3FFCD928 len 000126D8 (73 KiB): DRAM
> I (851) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (857) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (863) heap_init: At 4009E0EC len 00001F14 (7 KiB): IRAM
> I (871) spi_flash: detected chip: generic
> I (874) spi_flash: flash io: dio
> W (878) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
> W (889) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
> I (900) coexist: coex firmware version: 77cd7f8
> I (906) main_task: Started on CPU0
> I (916) main_task: Calling app_main()
> Starting r4sGate...
> I (916) R4S: Init start free heap: 206660
> 
> APP / IDF version: 2024.02.10 / v5.2-rc1-dirty
> Last reset reason: Software(3)
> I (1026) wifi:wifi driver task: 3ffd1eb0, prio:23, stack:6656, core=0
> I (1036) wifi:wifi firmware version: cc1dd81
> I (1036) wifi:wifi certification version: v7.0
> I (1036) wifi:config NVS flash: enabled
> I (1036) wifi:config nano formating: disabled
> I (1046) wifi:Init data frame dynamic rx buffer num: 24
> I (1046) wifi:Init static rx mgmt buffer num: 5
> I (1056) wifi:Init management short buffer num: 32
> I (1056) wifi:Init dynamic tx buffer num: 24
> I (1066) wifi:Init static rx buffer size: 1600
> I (1066) wifi:Init static rx buffer num: 6
> I (1066) wifi:Init dynamic rx buffer num: 24
> I (1076) wifi_init: rx ba win: 6
> I (1076) wifi_init: tcpip mbox: 32
> I (1086) wifi_init: udp mbox: 6
> I (1086) wifi_init: tcp mbox: 6
> I (1086) wifi_init: tcp tx win: 5744
> I (1096) wifi_init: tcp rx win: 5744
> I (1096) wifi_init: tcp mss: 1440
> I (1106) wifi_init: WiFi IRAM OP enabled
> I (1106) wifi_init: WiFi RX IRAM OP enabled
> I (1136) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
> I (1216) wifi:mode : sta (e8:6b:ea:f6:6c:cc)
> I (1216) wifi:enable tsf
> I (1226) wifi:ioctl: enable dynamic CS algorithm
> E (27806) wifi:NAN WiFi stop
> I (27806) wifi:flush txq
> I (27806) wifi:stop sw txq
> I (27806) wifi:lmac stop hw txq
> I (27986) wifi:mode : sta (e8:6b:ea:f6:6c:cc)
> I (27986) wifi:enable tsf
> I (27996) wifi:ioctl: enable dynamic CS algorithm
> E (55566) wifi:NAN WiFi stop
> I (55566) wifi:flush txq
> I (55566) wifi:stop sw txq
> I (55566) wifi:lmac stop hw txq
> ets Jul 29 2019 12:21:46
> 

Дальше перезапуск и все с начала...

alutov, 2024-02-12:
Да уж, при первом старте после прошивки подробный лог выключен. Подумаю, как это поправить.
Судя по логу, шлюз не находит или же не может подключиться к точке доступа r4s с паролем 12345678.

nightmare04, 2024-02-12:
> Да уж, при первом старте после прошивки подробный лог выключен. Подумаю, как это поправить. Судя по логу, шлюз не находит или же не может подключиться к точке доступа r4s с паролем 12345678.

И все таки иногда полезно читать инструкцию полностью, а не по диагонали, про то что надо создать сеть как то пропустил, думаю можно закрывать🤪