alutov/ESP32-R4sGate-for-Redmond #213: Карточки чайника и мультиварки для HA

00svd00, 2024-07-19:
Сделал для себя, возможно кому-то будет полезно. Для мультиварки отложенный запуск переделан в запуск ко времени(от текущей даты\времени сервера рассчитывается время до наступления события старта и отправляется мультиварке. Если текущее время меньше установленного(т.е. если сейчас на часах 22:00, а мы ставим запуск на 23:00) - старт случится в эти сутки. Если больше(т.е. если сейчас на часах 22:00, а мы ставим запуск на 08:00) - на следующие). С мультиваркой также есть ещё одна особенность - стартует с задержкой примерно в 5 секунд, т.к. между установкой значений переменных пришлось добавить задержки.
~/.homeassistant/packages/redmond.yaml
```
input_datetime:
  smart_cooker_start_time:
    name: "Мультиварка. Время запуска программы"
    has_date: false
    has_time: true
  smart_cooker_work_time:
    name: "Мультиварка. Время работы"
    has_date: false
    has_time: true
input_boolean:
  smart_cooker_delayed_start:
    name: "Мультиварка. Отложенный запуск"
    initial: off
input_button:
  smart_cooker_start:
    name: "Мультиварка. Запуск программы"
    icon: mdi:pot-steam
  smart_cooker_stop:
    name: "Мультиварка. Стоп"
    icon: mdi:close-octagon
  smart_kettle_start:
    name: "Чайник. Кипячение"
    icon: mdi:kettle-steam
  smart_kettle_stop:
    name: "Чайник. Стоп"
    icon: mdi:kettle-off

input_number:
  smart_cooker_temp_input:
    name: "Мультиварка. Температура"
    initial: 100
    min: 0
    max: 230
    step: 5

script:
  smart_cooker_run:
    sequence:
    - service: number.set_value
      target:
        entity_id: number.r4s01_cooker_temp
      data:
        value: "{{ states('input_number.smart_cooker_temp_input') | int }}"
    - delay: "00:00:01"
    - service: number.set_value
      target:
        entity_id: number.r4s01_cooker_s_min
      data:
        value: "{{ state_attr('input_datetime.smart_cooker_work_time','minute') | int }}"
    - delay: "00:00:01"
    - service: number.set_value
      target:
        entity_id: number.r4s01_cooker_s_hour
      data:
        value: "{{ state_attr('input_datetime.smart_cooker_work_time','hour') | int }}"
    - delay: "00:00:01"
    - service: number.set_value
      target:
        entity_id: number.r4s01_cooker_d_min
      data:
        value: >
          {% if states('input_boolean.smart_cooker_delayed_start') == "on" %}
            {{  ((now().replace(hour=state_attr('input_datetime.smart_cooker_start_time', 'hour'), minute=state_attr('input_datetime.smart_cooker_start_time', 'minute')) - now()).seconds%3600/60) | round(0, 'floor') }}
          {% else %}
            0
          {% endif %}
    - delay: "00:00:01"
    - service: number.set_value
      target:
        entity_id: number.r4s01_cooker_d_hour
      data:
        value: >
          {% if states('input_boolean.smart_cooker_delayed_start') == "on" %}
            {% if  now() - now().replace(hour=state_attr('input_datetime.smart_cooker_start_time', 'hour'), minute=state_attr('input_datetime.smart_cooker_start_time', 'minute')) > timedelta(seconds=60) %}
              {{ ((now().replace(hour=state_attr('input_datetime.smart_cooker_start_time', 'hour'), minute=state_attr('input_datetime.smart_cooker_start_time', 'minute')) + timedelta(days=1) - now()).seconds/3600) | round(0, 'floor')}}
            {% else %}
              {{  ((now().replace(hour=state_attr('input_datetime.smart_cooker_start_time', 'hour'), minute=state_attr('input_datetime.smart_cooker_start_time', 'minute')) - now()).seconds/3600) | round(0, 'floor') }}
            {% endif %}
          {% else %}
            0
          {% endif %}
    - delay: "00:00:01"
    - service: switch.turn_on
      target:
        entity_id: switch.r4s01_cooker_switch
automation:
  - alias: "Smart cooker start"
    trigger:
      platform: state
      entity_id: input_button.smart_cooker_start
    action:
      - service: script.smart_cooker_run
  - alias: "Smart cooker stop"
    trigger:
      platform: state
      entity_id: input_button.smart_cooker_stop
    action:
      - service: switch.turn_off
        target:
          entity_id: switch.r4s01_cooker_switch
  - alias: "Smart kettle start"
    trigger:
      platform: state
      entity_id: input_button.smart_kettle_start
    action:
      - service: switch.turn_on
        target:
          entity_id: switch.r4s02_kettle_boil
  - alias: "Smart kettle stop"
    trigger:
      platform: state
      entity_id: input_button.smart_kettle_stop
    action:
      - service: switch.turn_off
        target:
          entity_id: switch.r4s02_kettle_boil
```
Карточка чайника:
```
      - type: vertical-stack
        cards:
          - type: horizontal-stack
            cards:
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: input_button.smart_kettle_start
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: input_button.smart_kettle_stop
          - type: entity
            entity: water_heater.r4s02_kettle_wh_temp
          - type: thermostat
            entity: climate.r4s02_kettle_temp
          - type: horizontal-stack
            cards:
              - type: entity
                entity: sensor.r4s02_kettle_volume
              - type: entity
                entity: sensor.r4s02_kettle_volumelast
          - type: horizontal-stack
            cards:
              - type: entities
                entities:
                  - entity: switch.r4s02_kettle_heat
                  - entity: switch.r4s02_kettle_beep
                  - entity: switch.r4s02_kettle_backlight
                  - entity: light.r4s02_kettle_light
              - type: entities
                entities:
                  - entity: sensor.r4s02_kettle_energy
                  - entity: sensor.r4s02_kettle_ontime
                  - entity: sensor.r4s02_kettle_oncount
                  - entity: sensor.r4s02_kettle_rssi
```
Карточка мультиварки:
```
      - type: vertical-stack
        cards:
          - type: horizontal-stack
            cards:
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: input_button.smart_cooker_start
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: input_button.smart_cooker_stop
          - type: entity
            entity: sensor.r4s01_cooker_state
          - type: gauge
            entity: input_number.smart_cooker_temp_input
            max: 230
            needle: true
            severity:
              green: 95
              yellow: 160
              red: 200
          - type: entities
            entities:
              - entity: select.r4s01_cooker_program
              - entity: select.r4s01_cooker_auto_warming
                secondary_info: none
              - entity: input_datetime.smart_cooker_work_time
              - entity: input_boolean.smart_cooker_delayed_start
              - entity: input_datetime.smart_cooker_start_time
              - entity: sensor.r4s01_cooker_rssi
          - type: horizontal-stack
            cards:
              - type: entity
                entity: sensor.r4s01_cooker_hour
              - type: entity
                entity: sensor.r4s01_cooker_min
```

alutov, 2024-07-26:
Спасибо за информацию!

00svd00, 2024-10-03:
UPD
Оказывается с отложенным запуском мультиварки есть бага. Если сначала выбрать программу - отложенный запуск не срабатывает(соотв. срабатывает только в случае если сначала выставить время отложенного запуска, а затем уже выбирать программу и всё остальное). Поправил скрипт, теперь все параметры выставляются в правильном порядке, отложенный запуск работает