alutov/ESP32-R4sGate-for-Redmond #216: Wemos C3 mini (ESP32-C3FH4)

sviperz, 2024-10-08:
Плата [Wemos C3 mini](https://www.wemos.cc/en/latest/c3/c3_mini.html).

Прошил в нее файл *fr4sGate.bin* из папки **C3** командой:

`esptool --chip auto --port /dev/ttyACM0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x0 fr4sGateC3.bin`

Плата стартует и пытается подключиться к точке доступа, но у нее это по какой-то причине не получается. Точку доступа (SSID "r4s" без кавычек, пароль 12345678) создавал и на роутере и на телефоне. Телефон без проблем подключается к точке доступа на роутере, плата - ни в какую.

В чем может быть проблема?

Логи загрузки платы:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038076a
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2dac
entry 0x403ce000
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
I (24) boot: compile time 10:13:20
I (24) boot: chip revision: 3
I (25) boot.esp32c3: SPI Speed      : 40MHz
I (30) boot.esp32c3: SPI Mode       : DIO
I (35) boot.esp32c3: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (63) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (71) boot:  2 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  3 ota_0            OTA app          00 10 00010000 001f0000
I (86) boot:  4 ota_1            OTA app          00 11 00200000 001f0000
I (93) boot:  5 spiffs           Unknown data     01 82 003f0000 00010000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c140020 size=63fc8h (409544) map
I (199) esp_image: segment 1: paddr=00073ff0 vaddr=3fc95400 size=0316ch ( 12652) load
I (203) esp_image: segment 2: paddr=00077164 vaddr=40380000 size=08eb4h ( 36532) load
I (214) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=13c190h (1294736) map
I (485) esp_image: segment 4: paddr=001bc1b8 vaddr=40388eb4 size=0c434h ( 50228) load
I (504) boot: Loaded app from partition at offset 0x10000
I (504) boot: Disabling RNG early entropy source...
I (515) cpu_start: Unicore app
I (524) cpu_start: Pro cpu start user code
I (524) cpu_start: cpu freq: 160000000 Hz
I (525) app_init: Application information:
I (527) app_init: Project name:     r4sGate
I (532) app_init: App version:      1
I (536) app_init: Compile time:     Jul 28 2024 07:57:40
I (542) app_init: ELF file SHA256:  b5994440dde7b342...
I (548) app_init: ESP-IDF:          v5.3
I (553) efuse_init: Min chip rev:     v0.3
I (558) efuse_init: Max chip rev:     v1.99 
I (563) efuse_init: Chip rev:         v0.3
I (568) heap_init: Initializing. RAM available for dynamic allocation:
I (575) heap_init: At 3FCA2F20 len 0001D0E0 (116 KiB): RAM
I (581) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (588) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (595) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (603) spi_flash: detected chip: generic
I (606) spi_flash: flash io: dio
W (610) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (620) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (632) coexist: coex firmware version: dab85ae96
I (637) coexist: coexist rom version 9387209
I (642) main_task: Started on CPU0
I (642) main_task: Calling app_main()
Starting r4sGate...
I (642) R4S: Init start free heap: 237636

APP / IDF version: 2024.07.28 / v5.3
Last reset reason: Software(3)
I (682) pp: pp rom version: 9387209
I (682) net80211: net80211 rom version: 9387209
I (692) wifi:wifi driver task: 3fcabe74, prio:23, stack:6656, core=0
I (692) wifi:wifi firmware version: 0caa81945
I (692) wifi:wifi certification version: v7.0
I (692) wifi:config NVS flash: enabled
I (702) wifi:config nano formating: disabled
I (702) wifi:Init data frame dynamic rx buffer num: 24
I (702) wifi:Init static rx mgmt buffer num: 5
I (712) wifi:Init management short buffer num: 24
I (712) wifi:Init dynamic tx buffer num: 24
I (722) wifi:Init static tx FG buffer num: 2
I (722) wifi:Init static rx buffer size: 1600
I (732) wifi:Init static rx buffer num: 6
I (732) wifi:Init dynamic rx buffer num: 24
I (732) wifi_init: rx ba win: 6
I (742) wifi_init: accept mbox: 6
I (742) wifi_init: tcpip mbox: 32
I (742) wifi_init: udp mbox: 6
I (752) wifi_init: tcp mbox: 6
I (752) wifi_init: tcp tx win: 5744
I (762) wifi_init: tcp rx win: 5744
I (762) wifi_init: tcp mss: 1440
I (762) wifi_init: WiFi IRAM OP enabled
I (772) wifi_init: WiFi RX IRAM OP enabled
I (842) phy_init: phy_version 1180,01f2a49,Jun  4 2024,16:34:25
I (882) wifi:mode : sta (60:55:f9:28:04:78)
I (882) wifi:enable tsf
I (3552) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (3552) wifi:state: init -> auth (0xb0)
I (4552) wifi:state: auth -> init (0x200)
I (4562) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (9392) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (9392) wifi:state: init -> auth (0xb0)
I (10392) wifi:state: auth -> init (0x200)
I (10392) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (15232) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (15232) wifi:state: init -> auth (0xb0)
I (16232) wifi:state: auth -> init (0x200)
I (16232) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (21072) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (21072) wifi:state: init -> auth (0xb0)
I (22072) wifi:state: auth -> init (0x200)
I (22072) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (26912) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (26912) wifi:state: init -> auth (0xb0)
I (27912) wifi:state: auth -> init (0x200)
I (27912) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (32752) wifi:flush txq
I (32752) wifi:stop sw txq
I (32752) wifi:lmac stop hw txq
I (32772) wifi:mode : sta (60:55:f9:28:04:78)
I (32772) wifi:enable tsf
I (59682) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (59682) wifi:state: init -> auth (0xb0)
I (60682) wifi:state: auth -> init (0x200)
I (60682) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (60692) R4S: Failed to connect to SSID:'r4s', password:'12345678'
I (60692) R4S: Restarting now...
I (61692) wifi:flush txq
I (61692) wifi:stop sw txq
I (61692) wifi:lmac stop hw txq
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038076a
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2dac
entry 0x403ce000
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
...
```

sviperz, 2024-10-09:
@DmytroMelnyk может быть вы что-то подскажете?

sviperz, 2024-10-10:
Кажется, нашел причину. В [документации](https://www.wemos.cc/en/latest/c3/c3_mini_1_0_0.html) на эту плату указано, что в ревизии 1.0.0 (у меня как раз такая) для использования WiFi нужно выставлять мощность передатчика принудительно:

[image]

Проверил в MicroPython - действительно, без задания мощности плата не подключается, а если выставить мощность, то все OK:

[image]

@alutov, прошу добавить поддержку этой платы с такой особенностью WiFi.

sviperz, 2024-10-11:
Разобрался. Захардкодил с помощью `esp_wifi_set_max_tx_power(36)` и собрал. Все работает.

Плата [оказалась](https://github.com/arendst/Tasmota/discussions/15443#discussioncomment-2755046) со схемотехническим просчетом, это касается только версии 1.0.0, v2.1.0 не имеет таких проблем.

alutov, 2024-10-12:
Сегодня добрался до гита, а тут такой детектив). Но все равно ничего посоветовать бы не смог. В принципе, если сильно нужно, можно добавить мошность в настройках. Я пока с мощностью не разбирался, нужно почитать документацию, что там идет по умолчанию..

sviperz, 2024-10-12:
Главное, что с вами все ОК и вы на связи!) 🤝

Зато хоть и не без труда, но смог сам осилить фикс и сборку, получил интересный опыт.

Мощность в веб морде шлюза? До нее еще нужно добраться будет, а без нее плата не подключится к WiFi. Можно было бы поправить это в коде только для этой платы, если такое возможно. На других платах такой проблемы нет.

alutov, 2024-10-13:
Надеюсь,при сборке исрользовалась версия идф 5.3?

sviperz, 2024-10-13:
Где-то тут в тикетах увидел ([тут](https://github.com/alutov/ESP32-R4sGate-for-Redmond/pull/89#issuecomment-1313876912) и [тут](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/90#issuecomment-1192274663)), что надо использовать @5.2.0 или @5.0.0. Попробовал и получил такое:

[image]

Поэтому собирал с последним @6.9.0 и оно собралось.

Сейчас попробовал поставить @5.3.0, ошибок зависимостей нет. Лучше пересобрать на этой версии?

alutov, 2024-10-13:
Не работал с платформио, не подскажу, но желательно, чтобы после сборки прошивка показывала в вебе версию идф 5.3. Именно последняя версия шлюза и именно на этом идф работает стабильно и без перезагрузок.

sviperz, 2024-10-13:
Показывает вот что:

[image]

Значит, видимо, все в порядке.

alutov, 2024-10-13:
Ну да, уже вышла версия 5.3.1. Надеюсь, не хуже 5.3.)))

alutov, 2024-10-13:
Почитал по диагонали ссылочку. Как я понял, там предлагают убрать с платы одну емкость. И пишут, что уменьшение мощности не очень хорошее решение.

sviperz, 2024-10-13:
Да, я уже и сам хотел убрать эту емкость, когда не получалось собрать билд с фиксом. Но потом получилось и работает вроде сносно. Пока не буду трогать саму плату.

sviperz, 2024-10-17:
Думаю, статистика по этой плате сносная. 4 дня без ребутов, ошибок нет, работает стабильно.

[image]

sviperz, 2025-04-12:
@alutov не могли бы вы добавить поддержку этой платы или может просто собрать билд под нее?

Раньше мне удавалось добавить эту строчку, которая задает мощность Wi-Fi передатчика на этой плате (иначе Wi-Fi не работает, [здесь](https://www.wemos.cc/en/latest/c3/c3_mini_1_0_0.html) подробности) и собрать проект:

`esp_wifi_set_max_tx_power(36)`

Сейчас заметил у себя вот эту проблему https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/219 и хотел снова собрать с помощью PlatformIO, чтобы обновиться, но теперь не получается - компилятор выдает ошибку.