alutov/ESP32-R4sGate-for-Redmond #219: Постоянное переподключение wifi

EvgeniiMaslov, 2025-01-21:
Добрый день! Использую ATOM-LITE-ESP32, постоянно переподключается wifi, можно что то с этим сделать?

[image]
[image]

alutov, 2025-01-21:
Можно попробовать выбрать "802.11b/g/n" вместо "802.11b/g/n+dyn_cs".

EvgeniiMaslov, 2025-01-21:
> Можно попробовать выбрать "802.11b/g/n" вместо "802.11b/g/n+dyn_cs".

да я пробовал все варианты - не помогает((

alutov, 2025-01-21:
Можно еще снять птичку с "Reconnect if Mqtt loss". Ну и еще посмотреть debug лог по usb-uart.

EvgeniiMaslov, 2025-01-22:
> Можно еще снять птичку с "Reconnect if Mqtt loss". Ну и еще посмотреть debug лог по usb-uart.

снятие галки помогло, уже час без реконнектов и связь с устройством в mqtt не потерялась. Спасибо! А можно инструкцию как этот самый лог вытащить?

alutov, 2025-01-22:
Установить птичку "Uart Debug в настройках", подключить есп к компьютеру как при прошивке, запустить терминал, выставить параметры: скорость 11520, 8 бит, 1 стоп без проверки четности. И далее смотреть. Будет время, гляну еще, почему могут быть реконнекты из-за мктт. У меня реконнестов нет, хотя галка стоит.

EvgeniiMaslov, 2025-01-22:
> Установить птичку "Uart Debug" в настройках, подключить есп к компьютеру как при прошивке, запустить терминал, выставить параметры: скорость 11520, 8 бит, 1 стоп без проверки четности. И далее [...]

спасибо! будет время попробую посмотреть. может влияет то, что у меня mesh настроен в сети?

alutov, 2025-01-22:
У меня тоже меш кинетик. Наверное, брокеры разные. У меня москито, он всегда возвращает посланные ему данные. Под него и писал код. Возможно, другие брокеры этого могут не делать, потому и срабатывает реконнект по мктт.

EvgeniiMaslov, 2025-01-22:
> Поправил. В папку test залил файлик. Можно попробовать его с включенной галкой "Reconnect if Mqtt loss". Вроде бы должно работать.

спасибо! обновился и вернул галку на место, понаблюдаю и отпишусь.

EvgeniiMaslov, 2025-01-23:
> Поправил. В папку test залил файлик. Можно попробовать его с включенной галкой "Reconnect if Mqtt loss". Вроде бы должно работать.

все работает, за ночь ни разу не отвалилось!

closed 2025-01-26 by alutov