alutov/ESP32-R4sGate-for-Redmond #224: HLK-LD2450

Svet98, 2025-03-09:
Андрей здравствуйте!
Подскажите, датчик присутствия HLK-LD2450 есть в планах ввести?
Именно как датчик присутствия.
Спасибо.

alutov, 2025-03-09:
Пока не думал. Он чем-то лучшк 2410?

Svet98, 2025-03-09:
Говорят умеет одновременно "видеть" несколько человек не зависимо

Svet98, 2025-03-09:
вопросик по 2410, только подключила, без настроек, просто питание подкинула +5 от быстрой зарядки. Постоянно отваливается. Приходится ребутить или esp или сам датчик. Не подскажите по настройкам?
Спасибо.

Svet98, 2025-03-11:
сегодня немного "поигралась" с 2410 и 2450. Первый надо рядом с шлюзом держать, иначе постоянно отваливается. 2450 ловит из соседней комнаты. У него антенна внешняя.
Андрюш, может найдёте время для интеграции 2450?)
Спасибо!